mikakaraila/node-red-contrib-opcua #595: time problem with nodered and s7 opc-ua server -> error

DopplRaph, 2023-08-25:
Hello

I have a Siemens S7-1500 with OPC-UA-Server and this controller is connected to NodeRED.
I sucessfully connected the S7 with OPC-UA node.
[image]

If the clocktime on the server or client is different (e.g. 5 minutes - time of S7 is 08:20 and time of nodered is 08:25), the connection breaks down. I can´t connect to the server. Username and Password are correct. 
I get the error "Invalid endpoint, check that server has security policy: ["http://opcfoundation.org./UA/SecurityPolicy#Basic256Sha256"]"
This error does not indicate the real problem I think.
[image]
[image]

But with UaExpert or KepServerEX I can connect, also if the clocktime of server and client is different. So this problem is only in NodeRED.

I know that the solution is to set the same time on server and client.
But in our factory we have many machines from different manufacturers and the correct plc-time is not set everywhere. Sometimes the time is only a few minutes different to nodered and then we get the error in nodered. We have no timeserver.
I think this problem is general and not only with Siemens controls.

Any ideas?

NodeRED version v3.0.2
[image]

node-red-contrib-opcua v0.2.312
[image]

node version v20.5.1
[image]

configuration of the opc-ua node (automatic certificate)
[image]
[image]

Thank you and best regards
Raphael

CA1458, 2023-08-25:
> Certificate use needs time difference less than 1min.
> 
> I expect this should be found also from some basic OPC UA certificate use case documentation.

Thanks for your reply.
But is use certificate: None

mikakaraila, 2023-08-25:
Username and password are always cryptid with certificate

CA1458, 2023-08-25:
> 

When I use "Anonymous" the time is not relevant and it works?

mikakaraila, 2023-08-25:
I expect node-opcua will verify & validate certificate strictly, it will prevent all illegal use like time difference is too big.

CA1458, 2023-08-28:
> I expect node-opcua will verify & validate certificate strictly, it will prevent all illegal use like time difference is too big.

But if I use "Anonymous" it works, also when the times are different?

mikakaraila, 2023-08-28:
Anonymous is not using username/password which are always encrypted.  Encryption will validate/verify time & certificates.

CA1458, 2023-08-28:
> 

Ok so if I take anonymous without username/password it will work with time difference of server and client?

mikakaraila, 2023-08-28:
Yes, so I expect.

closed 2023-09-06 by CA1458