mikakaraila/node-red-contrib-opcua #596: Getting raw value at initial deployment

shubhamnodered, 2023-08-26:
Hi @mikakaraila,

I am using node-red to subscribe value from "Prosys OPCUA server".

My question is, **Why I'm also getting _raw value_** along with OPCUA server node value when i deploy the flow ?

It should be noted that this raw value only appears when i deploy the flow but further; if i make changes in opcua server, then no such raw value appears. Only desired changed value is returned in node-red.

raw value is always equals to  => **3234512922**

[image]

mikakaraila, 2023-08-27:
Your inject node will send message on initialization immediately that will most probably cause this.
Attach flow I can look deeper...

shubhamnodered, 2023-09-05:
@mikakaraila , I am attesting my flow here. But do note that this situation is particular to extraction of data from **"Prosys OPCUA server "** into node-red using **subscription** feature only. But it shouldn't be happening anyway.

`[
    {
        "id": "be296659c2391be1",
        "type": "tab",
        "label": "Flow 2",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "2b7aba71d75d9f75",
        "type": "debug",
        "z": "be296659c2391be1",
        "name": "debug 9",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 1100,
        "y": 220,
        "wires": []
    },
    {
        "id": "f002caad4803e580",
        "type": "OpcUa-Item",
        "z": "be296659c2391be1",
        "item": "ns=3;i=1002",
        "datatype": "Int32",
        "value": "",
        "name": "",
        "x": 520,
        "y": 220,
        "wires": [
            [
                "7a230f01843bc9c2"
            ]
        ]
    },
    {
        "id": "7a230f01843bc9c2",
        "type": "OpcUa-Client",
        "z": "be296659c2391be1",
        "endpoint": "27d85fe238a60af8",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": "500",
        "timeUnit": "ms",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "folderName4PKI": "",
        "useTransport": false,
        "maxChunkCount": 1,
        "maxMessageSize": 8192,
        "receiveBufferSize": 8192,
        "sendBufferSize": 8192,
        "name": "OPCUA client subscription",
        "x": 820,
        "y": 220,
        "wires": [
            [
                "2b7aba71d75d9f75"
            ]
        ]
    },
    {
        "id": "6c6835951b8a9133",
        "type": "comment",
        "z": "be296659c2391be1",
        "name": "Subscription interval must always be set at 500ms or less to avoid delay",
        "info": "",
        "x": 730,
        "y": 80,
        "wires": []
    },
    {
        "id": "5019060dfadfb2ee",
        "type": "inject",
        "z": "be296659c2391be1",
        "name": "",
        "props": [],
        "repeat": "",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "x": 290,
        "y": 220,
        "wires": [
            [
                "f002caad4803e580"
            ]
        ]
    },
    {
        "id": "27d85fe238a60af8",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://LAPTOP-7V1TBAAP:53530/OPCUA/SimulationServer",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

mikakaraila, 2023-09-07:
If you look msg topic it is subscriptionId not the nodeId!

closed 2023-09-09 by shubhamnodered