mikakaraila/node-red-contrib-opcua #598: Difference between frontend and backend

TheeeEnding, 2023-09-05:
Hello, I have the issue that the frontend UI shows an active subscription and doesn't get any new data/debug messages (see in second screenshot) even though there is a session timeout visible in the logs.

backend (the server time (13:02) corresponds to the actual time (15:02)):
[image]

frontend (last new debug msg at 15:02:11, screenshot was taken later while the logs showed the session timeout):
[image]

For the sake of completeness, this is the used flow:

`[
   {
       "id": "953786ea0d526a36",
       "type": "tab",
       "label": "Flow 2",
       "disabled": false,
       "info": "",
       "env": []
   },
   {
       "id": "6ee0f95658ba46c6",
       "type": "inject",
       "z": "953786ea0d526a36",
       "name": "",
       "props": [
           {
               "p": "topic",
               "vt": "str"
           }
       ],
       "repeat": "",
       "crontab": "",
       "once": false,
       "onceDelay": 0.1,
       "topic": "ns=0;i=2258",
       "x": 350,
       "y": 280,
       "wires": [
           [
               "fe86a797ed7738f7"
           ]
       ]
   },
   {
       "id": "fe86a797ed7738f7",
       "type": "OpcUa-Client",
       "z": "953786ea0d526a36",
       "endpoint": "d13c9839ccbe9b28",
       "action": "subscribe",
       "deadbandtype": "a",
       "deadbandvalue": 1,
       "time": 10,
       "timeUnit": "s",
       "certificate": "n",
       "localfile": "",
       "localkeyfile": "",
       "securitymode": "None",
       "securitypolicy": "None",
       "folderName4PKI": "",
       "useTransport": false,
       "maxChunkCount": 1,
       "maxMessageSize": 8192,
       "receiveBufferSize": 8192,
       "sendBufferSize": 8192,
       "name": "",
       "x": 580,
       "y": 280,
       "wires": [
           [
               "6df35ea0791235e0"
           ]
       ]
   },
   {
       "id": "6df35ea0791235e0",
       "type": "debug",
       "z": "953786ea0d526a36",
       "name": "debug 4",
       "active": true,
       "tosidebar": true,
       "console": false,
       "tostatus": false,
       "complete": "false",
       "statusVal": "",
       "statusType": "auto",
       "x": 800,
       "y": 280,
       "wires": []
   },
   {
       "id": "d13c9839ccbe9b28",
       "type": "OpcUa-Endpoint",
       "endpoint": "opc.tcp://10.149.58.140:4841",
       "secpol": "None",
       "secmode": "None",
       "none": true,
       "login": false,
       "usercert": false,
       "usercertificate": "",
       "userprivatekey": ""
   }
]`

The frontend basically doesn't update or show a timeout or keepalive while the backend is stating a session timout. What could be the reason for that? Are you familiar with this issue?

Thanks a lot in advance!

mikakaraila, 2023-09-05:
It is internal warning that comes from the node-opcua, perhaps @erossignon should comment this one.

TheeeEnding, 2023-09-05:
Mhm okay, for me it isn't so much about the warning but rather why your opcua client is showing "active subscribed" when the logs show that isn't

mikakaraila, 2023-09-05:
Code sets node status itself and warning is produced by node-opcua.

TheeeEnding, 2023-09-06:
Alright, first of all thanks for the fast answers... however I think I'm still not really getting the point, sorry I'm quite new to this.

So you're saying that your code can't recognize/see the warning(s) that are produced by node-opcua and therefore is not able to set the correct status of the node?

mikakaraila, 2023-09-07:
Yes, but retest with the latest version as there is new version of my code and node-opcua now available.

TheeeEnding, 2023-09-08:
Alright, thank you! I'll get back to you when I have news

TheeeEnding, 2023-09-14:
I can't reproduce the failure, so I think the update fixed it, thank you!

closed 2023-09-14 by TheeeEnding