mikakaraila/node-red-contrib-opcua #599: Automatic reconnect doesn't work (same setup with a seperate node.js-client works)

fanbel, 2023-09-05:
Hi Mika!
Currently we're experiencing a problem regarding the reconnection based on a connection loss between the node-red-opcua-client and the respective opcua server, be it because of a network connection loss or because of an outage of the server.

E.g. the situation looks like this: 
- We have a stable connection (subscription) from our node-red-opcua-client to the respective opcua server. That means the node states "active subscribing" and gets data, if new data gets published from the server.
- Now the connection gets cut/lost, e.g. the server goes offline.
- After a while (it doesn't matter how long the outage is) the server goes online again.
- We would expect that after the server is online again, new data gets retrieved by the opcua client, but that isn't the case, as we don't get anything. In the logs I found that the following "error": client_session_keepalive_manag:130    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed
- The strange thing is that we built a seperate opcua client based on node.js: and the reconnect with the same setup works with this client

Our "system" data:
- node red is run via kubernetes as an underlying docker image
- node red version 3.0.2
- node red opcua client version 0.2.310
- node js version ^2.108.0

I hope you can help us, by answering the summarized question: "Why does the reconnect work with the node.js-client but not with the node red opcua client?"

If you need further logs, flows or anything, let me know.

Thank you!

PS: After the above mentioned test, we tried to fix the problem by using newer versions of your client (including the underlying newer node.js client), sadly it didn't work.
Also because it works with node.js we didn't pursue further communication with Etienne.

mikakaraila, 2023-09-05:
I have to verify this with @erossignon cause there is internal keepalive subscription and node-red-contrib-opcua sets also keepalive. So there may be now two subscriptions with keepalive.

fanbel, 2023-09-05:
okay thanks for replying and I'm looking forward to hearing from you afterwards!

mikakaraila, 2023-09-07:
Please re-test with new just published version.

fanbel, 2023-09-08:
thank you, it did work now.
what exactly was the root cause for this?

mikakaraila, 2023-09-08:
I was using keepalive true with session and it seems to be so that node-opcua creates own keepalive subscription automatically and these together will not work.

closed 2023-09-08 by mikakaraila

RedShift1, 2023-10-17:
I'm still having the same problem. "Invalid Channel BadConnectionClosed", version 0.2.316.

RedShift1, 2023-10-18:
Downgrading to 0.2.292 didn't work...