mikakaraila/node-red-contrib-opcua #601: DEADBAND TYPE AND VALUE IN MONITOR

carlosbama3, 2023-09-07:
Good morning Mika, I'm doing some test about to set a deadband type and value in monitor.

For example, I set this configuration as an input for opc ua client:

[image]
 And I have get result as output, but I should only have results when the difference between the last result and the next is more than the deadband value.

I only obtain one result and then the state of opc ua client is keepalive. But it should be more than 1 result, because there are many results that thet are over that the difference.

Thanks for all,

Carlos

mikakaraila, 2023-09-07:
I expect deadband absolute/percent is meant to be used only with scalars not with arrays.
As scalar variable can have EUrange to look & use for percent.

carlosbama3, 2023-09-07:
But if i set these fixed parameters in OPC UA Client, there is not any problem. It works well. Independently if the tag is array or not

In my case, I want to establish for each node a different deadband type and value. Can I do this?

mikakaraila, 2023-09-07:
Yes you can., but at the moment deadband is fixed - I mean you cannot inject new deadband parameters for the client node.
Deadband 500 means that the initial value is 100, you will get update when value is 601 or -401.

carlosbama3, 2023-09-07:
Okey thanks. But I don't understand if i can do it or not.

Let me explain better:
Imagine I have two topics or tags tag1 and tag2. I want to pass different parameters to the OPC UA client node to monitor each tag differently.

That is to say,
I want tag1 to have an interval time of 300 seconds, an absolute deadbandType and a deadbandValue of 500. I want tag2 to have an interval time of 300 seconds but an absolute deadbandType and a deadbandValue of 300.

What is clear to me is that it doesn't matter if it is scalar or array, if it is fixed, the deadband will work. But in my case, I want to pass different types of deadband to the client node.

Is this not possible?

Thanks

mikakaraila, 2023-09-07:
As I wrote deadband is fixed - it means you can use absolute with value like 1 for all items
 or 
percent with 1% => this will use each variable EUrange with 1%. 
If EUrange is from 0 to 100 value is 1 
if EUrange is 0 to 50 then value 0.5 etc.

mikakaraila, 2023-09-07:
Sorry I just read my code and you can actually inject msg with deadbandType

msg.deadbandType = "a"; // Absolute
msg.deadbandType = "p"; // Percent
msg.deadbandValue = 300; // Value to be used in DataChangeFilter

carlosbama3, 2023-09-07:
Okey Rhanks

mikakaraila, 2023-09-07:
Read my comment above!

carlosbama3, 2023-09-07:
Sorry, okey. But i don't understand what is my problem 

Be cause I inject these in opc ua client:

[image]

And the result is:

[image]
 
But later, the state of opc ua client and it doesn't send the next results

[image]

carlosbama3, 2023-09-08:
I don't know how to do to have the monitorization correctly.

mikakaraila, 2023-09-08:
I will make example... not today but as soon as I will have time.

carlosbama3, 2023-09-08:
Okey thanks, I have changed the type of values, becuse in the last picture you can see that all the values are strings, but for interval and the deadband Value are integers.

Thanks

carlosbama3, 2023-09-08:
But the problem remains

carlosbama3, 2023-09-11:
I don't know if I the parameter os each variable I should be within payload like this:

[image]

mikakaraila, 2023-09-11:
Under msg object as I wrote above:
msg.deadbandType = "a"; // Absolute
msg.deadbandType = "p"; // Percent
msg.deadbandValue = 300; // Value to be used in DataChangeFilter

carlosbama3, 2023-09-11:
and the same way for interval?

mikakaraila, 2023-09-11:
It can be inject as msg.payload. Default value is 100.

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila