mikakaraila/node-red-contrib-opcua #602: Unable to connect to endpoint

AlbertoPilatoDerga, 2023-09-08:
Hello Mika,
we are having issues connecting to an ADAM endpoint

[image]

We are able to connect using UaExpert or Prosys OPC UA Browser with None/None/Anonymous

[image]

Can you please advise us on how to troubleshoot this connection?

Regards and TIA,
Alberto

mikakaraila, 2023-09-09:
Are you using latest version?

AlbertoPilatoDerga, 2023-09-11:
Hi mika, yes, it's 0.2.314

[image]

mikakaraila, 2023-09-11:
As UaExpert uses anonymous I expect you are using same with node-red, can you provide client node endpoint dialog?
Just to see your settings...

mikakaraila, 2023-09-13:
Some cleanup, re-open or comment original issue.

closed 2023-09-13 by mikakaraila

AlbertoPilatoDerga, 2023-09-14:
[image]
[image]

mikakaraila, 2023-09-15:
Turn off (uncheck) transport settings, server cannot understand those most probably.

AlbertoPilatoDerga, 2023-09-15:
Sorry, but I'm getting the same errors even with "Use transport settings" unchecked.

[image]

mikakaraila, 2023-09-15:
Use empty string on pki folder setting

AlbertoPilatoDerga, 2023-09-15:
[image]
 
same situation :(

AlbertoPilatoDerga, 2023-09-19:
Hello mika, can you please advise us on how to proceed?

mikakaraila, 2023-09-19:
I expect you are using this as OPC UA Server:
https://www.integral-system.fr/shop/media/product/file/adam6300_user_manaul_ed2final-623b52efa2b97.pdf

[image]

mikakaraila, 2023-09-19:
Basic128Rsa15 seems to be obsolete (not secure enough anymore):
[image]

mikakaraila, 2023-09-19:
And this is also good to check:
[image]

AlbertoPilatoDerga, 2023-09-21:
Hi mika, we did not manage to connect, neither with None nor with Basic128Rsa15.

Also the check regarding BadSecureChannelClosed does not show up, we can connect without issues to the endpoint via ADAM Utility.

We're going to try integration via Modbus if no other options are available....

mikakaraila, 2023-09-21:
For me it seems that you haven´t accepted certificate / it cannot be used anymore for encrypting as it is obsolete.

And it is needed in first connection as documentation says:
[image]