mikakaraila/node-red-contrib-opcua #603: OpcUa Client fails to connect with the Server

simarsamra, 2023-09-08:
Hi
I am facing this strange issue with node-red-contrib-opcua node Client.
I have three OpcUa servers on my network. They all have the same configuration.
Using UaExpert, I can connect to all these servers without any issues.
But when I use node-red Opc Ua Client to connect out of three it only works with one server. For the remaining two servers I get an Invalid Endpoint error. The Endpoint node properties are set to exactly the same as in UaExpert. I searched the internet for the solution but could not find a fix.
Please take a look at the picture below.

Is this a known issue? How can I fix it?
Thank you in advance.
[image]

simarsamra, 2023-09-08:
#587

mikakaraila, 2023-09-09:
If one server works with same settings then have you trusted client certificate on those two not working ones?

closed 2023-09-09 by mikakaraila

simarsamra, 2023-09-20:
Hi Mika
The servers auto-accept the certificates. I use a username and password. Here is what I figured out till now.
I installed a VMware on my PC. Installed Node-Red on VMware(Windows 10). I was able to connect to all my servers without any issues from VMware,
But from my PC, I could not connect to two servers using Node-Red OpcUa Client Node.
I tried uninstalling NodeJS, and Node-Red. I cleared all the Node-Red-related folders from the AppData folder. 
What could be giving me this error? Any specific files in Windows that I can try to delete?
Thank you

[image]