mikakaraila/node-red-contrib-opcua #605: Can't instantiate

S-Carol, 2023-09-11:


mikakaraila, 2023-09-11:
I will close this one if you cannot provide any description...

S-Carol, 2023-09-11:
> I will close this one if you cannot provide any description...

Sorry, I accidentally opened the issue before finishing writing. I'll open a new one with the correct title and description

S-Carol, 2023-09-11:
> I will close this one if you cannot provide any description...

mikakaraila, 2023-09-11:
Can you provide simple flow and server to debug issue? @erossignon just fixed something related to ExtensionObjects.

The latest version should use fixed node-opcua, but perhaps I should change something...

S-Carol, 2023-09-11:
The flow and server are as in the [example](https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/examples/CUSTOM_OBJECT.json).
I wasn't sure if it was the same issue since this one is about adding Extension Objects, not reading or writing already existing ones. But they could be related.
I tried updating node-opcua, just in case it worked, but there was no change.

mikakaraila, 2023-09-12:
I tested and it works:
[image]

mikakaraila, 2023-09-12:
I expect your environment is missing nodeset or did you inject from up to down inject nodes?

xxyzz90, 2024-02-04:
Hi, I might be facing a similar Issue. I am running 0.2.321. My goal is to Create a OPCUA Server that holds an ExtensionObject for Simulation. 
I used the Example Flow

[flows.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/14155899/flows.json)

Debug form Inject node Gives me:
{"_msgid":"cf7b85248e105a2d","payload":{"opcuaCommand":"addVariable"},"topic":"ns=1;s=Test2;datatype=ExtensionObject;typeId=ns=2;i=3010"}

And the Error in Debug from OPCUA-Server Node:
Error: expecting a dataType

Thanks in Advnance