mikakaraila/node-red-contrib-opcua #610: Node-RED crashes when WRITING to OPC-UA Endpoint

AlbertoPilatoDerga, 2023-09-21:
Hello,
i'm writing these messages to the client node, but when I inject, it causes the Node-RED instance to crash.

[image]

Here is the error message

[image]

I'm on nodered/node-red:3.0.2-18 and node-red-contrib-opcua:0.2.314

mikakaraila, 2023-09-21:
opcua-item node msg object contains these fields:
[image]

mikakaraila, 2023-09-21:
And if you want to write multiple take a look examples OPCUA-TEST-NODES.json or NR-WEBINAR-MULTIPLE.json

mikakaraila, 2023-09-22:
Your flow or part of it would help me to find error...

AlbertoPilatoDerga, 2023-09-22:
Below you will finde the flow i'm executing. I've managed to understand that the crash is caused by writing values inadequate to the destination datatype (eg. a string in an Int16 field).

I tried to put a catch node, but it does not seem to work.

`[{"id":"f7eec4a4789a5fc1","type":"tab","label":"TecnoATS","disabled":false,"info":"","env":[]},{"id":"4510c68483ad0f18","type":"inject","z":"f7eec4a4789a5fc1","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":180,"y":80,"wires":[["34cf0188cdf006d5"]]},{"id":"9d583dbe9b55baf1","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 31","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":960,"y":220,"wires":[]},{"id":"96da2c1b78bca869","type":"function","z":"f7eec4a4789a5fc1","name":"function 3","func":"let outArray = [];\n\nfor (const property in msg.payload.payload) {\n    let suffix = 'o', dataType;\n    switch(property) {\n        case 'RicettaPLC':\n            suffix = 'a';\n            dataType = 'Int16'\n            break;\n        case 'RicettaRobopack':\n            dataType = 'Int16'\n            break;\n        case 'CodiceArticolo':\n            dataType = 'String'\n            break;\n        case 'DescrizioneArticolo':\n            dataType = 'String'\n            break;\n        case 'LayoutEtichetta':\n            dataType = 'Int16'\n            break;\n        case 'Ordine':\n            dataType = 'String'\n            break;\n        case 'PezziPallet':\n            dataType = 'Int32'\n            break; \n        case 'PezziScatola':\n            dataType = 'Int32'\n            break;\n        case 'PezziTotaliRichiesti':\n            dataType = 'Int32'\n            break;\n        default:\n            suffix = 'o';\n    }\n\n    outArray.push({\n        topic: 'ns=2;s=' + msg.payload.impianto_ats + '.Tags.' + property + '_successiv' + suffix,\n        datatype: dataType,\n        payload: msg.payload.payload[property]\n    });\n}\n\n//msg.outArray = outArray;\n\nreturn [ outArray ];","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":600,"y":220,"wires":[["0cee938e88c4d79f","7397b2677a801e5f"]]},{"id":"98d570d2ca73c5f1","type":"switch","z":"f7eec4a4789a5fc1","name":"Ins. / Agg.","property":"payload.azione","propertyType":"msg","rules":[{"t":"eq","v":"Inserimento","vt":"str"},{"t":"eq","v":"Aggiornamento","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":340,"y":280,"wires":[["96da2c1b78bca869"],[]]},{"id":"7397b2677a801e5f","type":"OpcUa-Client","z":"f7eec4a4789a5fc1","endpoint":"5988fc1f6abbfa8d","action":"write","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","useTransport":false,"maxChunkCount":"","maxMessageSize":"","receiveBufferSize":"","sendBufferSize":"","name":"","x":780,"y":220,"wires":[["9d583dbe9b55baf1"]]},{"id":"0cee938e88c4d79f","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 32","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":760,"y":300,"wires":[]},{"id":"34cf0188cdf006d5","type":"http request","z":"f7eec4a4789a5fc1","name":"","method":"GET","ret":"obj","paytoqs":"ignore","url":"http://10.203.128.120:8081/api/rest/getCodaTecnoATS","tls":"","persist":true,"proxy":"","insecureHTTPParser":false,"authType":"","senderr":false,"headers":[],"credentials":{},"x":350,"y":80,"wires":[["ac1ef7d9a06bbc5e"]]},{"id":"f1324e8bbac3e380","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 34","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":800,"y":80,"wires":[]},{"id":"ac1ef7d9a06bbc5e","type":"split","z":"f7eec4a4789a5fc1","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":510,"y":80,"wires":[["b7168ee8080e290c"]]},{"id":"b7168ee8080e290c","type":"split","z":"f7eec4a4789a5fc1","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":650,"y":80,"wires":[["98d570d2ca73c5f1","f1324e8bbac3e380"]]},{"id":"e7c66af379b3a208","type":"catch","z":"f7eec4a4789a5fc1","name":"","scope":null,"uncaught":false,"x":280,"y":380,"wires":[["aa81f52e81e513e6"]]},{"id":"aa81f52e81e513e6","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 35","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":560,"y":380,"wires":[]},{"id":"5988fc1f6abbfa8d","type":"OpcUa-Endpoint","endpoint":"opc.tcp://192.168.213.14:4840","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""}]`

AlbertoPilatoDerga, 2023-09-22:
The error is catched properly when using WRITE MULTIPLE instead of WRITE

`[
    {
        "id": "f7eec4a4789a5fc1",
        "type": "tab",
        "label": "TecnoATS",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "4510c68483ad0f18",
        "type": "inject",
        "z": "f7eec4a4789a5fc1",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 180,
        "y": 80,
        "wires": [
            [
                "34cf0188cdf006d5"
            ]
        ]
    },
    {
        "id": "9d583dbe9b55baf1",
        "type": "debug",
        "z": "f7eec4a4789a5fc1",
        "name": "debug 31",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 960,
        "y": 220,
        "wires": []
    },
    {
        "id": "96da2c1b78bca869",
        "type": "function",
        "z": "f7eec4a4789a5fc1",
        "name": "function 3",
        "func": "let outArray = [];\n\nfor (const property in msg.payload.payload) {\n    let suffix = 'o', dataType;\n    switch(property) {\n        case 'RicettaPLC':\n            suffix = 'a';\n            dataType = 'Int16'\n            break;\n        case 'RicettaRobopack':\n            dataType = 'Int16'\n            break;\n        case 'CodiceArticolo':\n            dataType = 'String'\n            break;\n        case 'DescrizioneArticolo':\n            dataType = 'String'\n            break;\n        case 'LayoutEtichetta':\n            dataType = 'Int16'\n            break;\n        case 'Ordine':\n            dataType = 'String'\n            break;\n        case 'PezziPallet':\n            dataType = 'Int32'\n            break; \n        case 'PezziScatola':\n            dataType = 'Int32'\n            break;\n        case 'PezziTotaliRichiesti':\n            dataType = 'Int32'\n            break;\n        default:\n            suffix = 'o';\n    }\n\n/*\n\n    outArray.push({\n        topic: 'ns=2;s=' + msg.payload.impianto_ats + '.Tags.' + property + '_successiv' + suffix,\n        datatype: dataType,\n        payload: msg.payload.payload[property]\n    });\n}\n\n//msg.outArray = outArray;\n\nreturn [ outArray ];\n\n*/\n\n    outArray.push({\n        nodeId: 'ns=2;s=' + msg.payload.impianto_ats + '.Tags.' + property + '_successiv' + suffix,\n        datatype: dataType,\n        value: msg.payload.payload[property]\n    });\n}\n\nmsg.topic = 'ns=2;s=' + msg.payload.impianto_ats + '.Tags'\nmsg.payload = outArray;\n\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 600,
        "y": 220,
[... 183 lines cut ...]
        "name": "debug 35",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 560,
        "y": 380,
        "wires": []
    },
    {
        "id": "5988fc1f6abbfa8d",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.213.14:4840",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

closed 2023-09-22 by AlbertoPilatoDerga

mikakaraila, 2023-09-22:
I expect you are missing something from your switch case that will produce msg object without datatype.
[image]

Can you provide flow that is not using REST API, instead just JSON file that will be fetched to function 3 node?

AlbertoPilatoDerga, 2023-09-22:
`{"payload":{"id":1,"azione":"Inserimento","impianto_ats":"297","payload":{"Ordine":"123","RicettaPLC":"MES_PLAST_MAT_CUST_DATA.ZZPLC","PezziPallet":10,"PezziScatola":1,"CodiceArticolo":"MPM_AFKOL.MATNR","LayoutEtichetta":"MES_PLAST_MAT_CUST_DATA.ZZLAYETIC","RicettaRobopack":"MES_PLAST_MAT_CUST_DATA.ZZROBOPACK","DescrizioneArticolo":"MPM_MATMAS_HDRT.MAKTX","PezziTotaliRichiesti":123},"timestamp_elaborato":null,"timestamp_inserimento":"2023-09-21T16:35:12.141563+02:00"},"topic":"","statusCode":200,"headers":{"transfer-encoding":"chunked","date":"Fri, 22 Sep 2023 09:48:26 GMT","server":"Warp/3.3.14","x-request-id":"6159bcef-c2fb-43a9-abc4-2b0b80bdb8fa","content-type":"application/json; charset=utf-8","x-node-red-request-node":"78b3373a"},"responseUrl":"http://10.203.128.120:8081/api/rest/getCodaTecnoATS","redirectList":[],"retry":0,"parts":{"parts":{"id":"b659917fc6448dcd","type":"object","key":"coda_tecnoats","index":0,"count":1},"id":"30b1ec419320ea39","type":"array","count":3,"len":1,"index":2},"_msgid":"aa2c537c2d519d3f"}`

mikakaraila, 2023-09-22:
Here you can see that datatype is not defined or msg object is not used correctly, please debug yourself more your flow:
[image]

mikakaraila, 2023-09-22:
I cannot test your flow, but made small snippet that you should try (at least it looks good for me):
[image]

mikakaraila, 2023-09-22:
Here is flow with addition:
`[{"id":"f7eec4a4789a5fc1","type":"tab","label":"TecnoATS","disabled":false,"info":"","env":[]},{"id":"5988fc1f6abbfa8d","type":"OpcUa-Endpoint","endpoint":"opc.tcp://192.168.213.14:4840","secpol":"None","secmode":"None","none":true,"login":false,"usercert":false,"usercertificate":"","userprivatekey":""},{"id":"4510c68483ad0f18","type":"inject","z":"f7eec4a4789a5fc1","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":180,"y":80,"wires":[["34cf0188cdf006d5"]]},{"id":"9d583dbe9b55baf1","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 31","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":960,"y":220,"wires":[]},{"id":"96da2c1b78bca869","type":"function","z":"f7eec4a4789a5fc1","name":"function 3","func":"let outArray = [];\n\nfor (const property in msg.payload.payload) {\n    let suffix = 'o', dataType;\n    switch(property) {\n        case 'RicettaPLC':\n            suffix = 'a';\n            dataType = 'Int16'\n            break;\n        case 'RicettaRobopack':\n            dataType = 'Int16'\n            break;\n        case 'CodiceArticolo':\n            dataType = 'String'\n            break;\n        case 'DescrizioneArticolo':\n            dataType = 'String'\n            break;\n        case 'LayoutEtichetta':\n            dataType = 'Int16'\n            break;\n        case 'Ordine':\n            dataType = 'String'\n            break;\n        case 'PezziPallet':\n            dataType = 'Int32'\n            break; \n        case 'PezziScatola':\n            dataType = 'Int32'\n            break;\n        case 'PezziTotaliRichiesti':\n            dataType = 'Int32'\n            break;\n        default:\n            suffix = 'o';\n    }\n\n/*\n\n    outArray.push({\n        topic: 'ns=2;s=' + msg.payload.impianto_ats + '.Tags.' + property + '_successiv' + suffix,\n        datatype: dataType,\n        payload: msg.payload.payload[property]\n    });\n}\n\n//msg.outArray = outArray;\n\nreturn [ outArray ];\n\n*/\n\n    outArray.push({\n        nodeId: 'ns=2;s=' + msg.payload.impianto_ats + '.Tags.' + property + '_successiv' + suffix,\n        datatype: dataType,\n        value: msg.payload.payload[property]\n    });\n}\n\nmsg.topic = 'ns=2;s=' + msg.payload.impianto_ats + '.Tags'\nmsg.payload = outArray;\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":600,"y":220,"wires":[["0cee938e88c4d79f","7397b2677a801e5f"]]},{"id":"98d570d2ca73c5f1","type":"switch","z":"f7eec4a4789a5fc1","name":"Ins. / Agg.","property":"payload.azione","propertyType":"msg","rules":[{"t":"eq","v":"Inserimento","vt":"str"},{"t":"eq","v":"Aggiornamento","vt":"str"}],"checkall":"true","repair":false,"outputs":2,"x":420,"y":240,"wires":[["96da2c1b78bca869"],[]]},{"id":"7397b2677a801e5f","type":"OpcUa-Client","z":"f7eec4a4789a5fc1","endpoint":"5988fc1f6abbfa8d","action":"writemultiple","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","folderName4PKI":"","useTransport":false,"maxChunkCount":"","maxMessageSize":"","receiveBufferSize":"","sendBufferSize":"","name":"","x":780,"y":220,"wires":[["9d583dbe9b55baf1"]]},{"id":"0cee938e88c4d79f","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 32","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":760,"y":300,"wires":[]},{"id":"34cf0188cdf006d5","type":"http request","z":"f7eec4a4789a5fc1","name":"","method":"GET","ret":"obj","paytoqs":"ignore","url":"http://10.203.128.120:8081/api/rest/getCodaTecnoATS","tls":"","persist":true,"proxy":"","authType":"","senderr":false,"x":350,"y":80,"wires":[["ac1ef7d9a06bbc5e"]]},{"id":"f1324e8bbac3e380","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 34","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":800,"y":80,"wires":[]},{"id":"ac1ef7d9a06bbc5e","type":"split","z":"f7eec4a4789a5fc1","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":510,"y":80,"wires":[["b7168ee8080e290c"]]},{"id":"b7168ee8080e290c","type":"split","z":"f7eec4a4789a5fc1","name":"","splt":"\\n","spltType":"str","arraySplt":1,"arraySpltType":"len","stream":false,"addname":"","x":650,"y":80,"wires":[["98d570d2ca73c5f1","f1324e8bbac3e380"]]},{"id":"e7c66af379b3a208","type":"catch","z":"f7eec4a4789a5fc1","name":"","scope":null,"uncaught":false,"x":280,"y":380,"wires":[["aa81f52e81e513e6"]]},{"id":"aa81f52e81e513e6","type":"debug","z":"f7eec4a4789a5fc1","name":"debug 35","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":560,"y":380,"wires":[]},{"id":"444e72c92b5d9675","type":"file in","z":"f7eec4a4789a5fc1","name":"REST MSG","filename":"REST-MSG4ISSUE-610.json","format":"utf8","chunk":false,"sendError":false,"encoding":"none","allProps":false,"x":350,"y":140,"wires":[["ac1ef7d9a06bbc5e","7a92f78e3ba584f4"]]},{"id":"ccbd240a2164470d","type":"inject","z":"f7eec4a4789a5fc1","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"" [... cut ...]        datatype: dataType,\n        value: msg.payload.payload.payload[property]\n    });\n}\nnode.send({topic: \"writemultiple\", payload: \"\"});","outputs":1,"noerr":0,"initialize":"","finalize":"","libs":[],"x":1000,"y":600,"wires":[["77c74bc915a531a2"]]},{"id":"77c74bc915a531a2","type":"debug","z":"f7eec4a4789a5fc1","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1150,"y":600,"wires":[]}]`