mikakaraila/node-red-contrib-opcua #611: Setting Source TimeStamp in my OPC UA server when writing a value

rknasr, 2023-09-25:
Hello everyone, 

I have deployed an OPC UA server using the OPC UA server node. Upon writing a value to a certain node in my OPC UA server, and assigning a source timestamp, there seems to be a possible bug. The source timestamp assigned is being written to the server timestamp instead. The value is written sucessfully to this node, but the source Timestamp is being written to the server timestamp. 
This is the current payload injectedto the opc ua server:
msg.payload= {"**namespace**": 10, 
"**messageType**": "Variable", 
"**variableName**":""Objects"."DeviceSet"."AssemblySystem01"."Assets"."Tools"."Value"",
  "**variableValue**": 47,
 "**sourceTimestamp**": 1688997600000,
    "**quality**": "Good"}
return msg;

Also below is the result when accessed with an OPC UA client:

[image]

mikakaraila, 2023-09-26:
I will take a look later...

mikakaraila, 2023-09-26:
OK, found it and fixed.

rknasr, 2023-09-27:
Hi mika, thanks for the quick fix!  I just tested the update now, it no longer changes just the server timestamp, but what's happening now is both the server timestamp and source time stamp are being changed.  So for example in the image below I injected the source timestamp to be 6 am, while the server timestamp shall remain 11 am( as the server timestamp shall always be the actual servers time, shouldn't be altered). 

[image]

rknasr, 2023-09-27:
Hi mika, thanks for the quick fix! I just tested the update now, it no longer changes just the server timestamp, but what's happening now is both the server timestamp and source time stamp are being changed. So for example in the image below I injected the source timestamp to be 6 am, while the server timestamp shall remain 12pm( as the server timestamp shall always be the actual servers time).
[image]

mikakaraila, 2023-09-27:
I am not setting serverTimestamp. 

API is  setValueFromSource(value: VariantLike, statusCode?: StatusCode, sourceTimestamp?: Date): void;

Server will set it if value changes or statusCode changes or sourceTimestamp changes.
So the current functionality is correct.

closed 2023-09-27 by mikakaraila

rknasr, 2023-10-04:
Ah I see, so I guess it's an API issue in the sdk then?  
So upon using the timestamp from my connected device as a source timestamp in my opc ua server, the client connected to the opc ua server will read both source and server time stamp to be identical. (where the server timestamp is set to be like the source timestamp as the snippet attached in the previous comment)
This actually raises some issues, so for example lets say my device isn't synchronized with an NTP , and therefore my source timestamp is wrong, this would result in having both my source timestamp and server timestamp of my opc ua server to be also wrong, making them both unreliable timestamps.
Could it be possible to raise an issue in the node opc ua sdk concerning this?