mikakaraila/node-red-contrib-opcua #612: Client set to history, time range?

Theodorska1, 2023-09-26:
Hello, 
I am trying to read the max value of a node over a specified period of time. 
Is it possible to set the history range in time on a client set to HISTORY?

I am currently just sending this message via a function node into the client set to history:

msg.topic ="ns=4;s=API.CCS1_ST.ChargingCurrent"
msg.aggregate = "max";
return msg;

and the returning payload from the client is just a status code with a value that looks like a random super high number that is not reflecting any real max value from this sender.

mikakaraila, 2023-09-26:
Take a look into OPCUA-TEST-NODES.json, I don´t know what you want/try to do, but check this first:
[image]

Theodorska1, 2023-09-27:
Thank you mister!
That's exactly what I was looking for:

"msg.topic="ns=1;s=TestAddVariable3";
msg.aggregate="max";
msg.end = Date.now() + (2 * 60 * 60); // GMT + 2h
msg.start = 1638132240000; //  msg.end - (6 * 60 * 60); // 10 min
return msg;;"

You have done an amazing job!

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila