mikakaraila/node-red-contrib-opcua #613: invalid endpoint after debugging

Theodorska1, 2023-10-03:
Hello, 

I use one client set to subscribe where i delete the subscription for every redeployment and debugging.
I also have several client nodes used to read on trigger, and one to write.  I also use one browse.

My problem is after some time working on my project and debugging i end up getting "Invalid Endpoint. check that server has a security policy..." When this error occurs none of the OPC nodes are able to reconnect. and if I try to reploy or trigger an opc node I get these error messages in this order:

"OpcUaClientNode: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [3]"

"OpcUaClientNode:         using securityPolicy: ["http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256"]"

"OpcUaClientNode: Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: XXX and password: XXX"

"OpcUaClientNode: Case C: With Sign you cannot use SecurityPolicy None!!"

"OpcUaClientNode: Invalid endpoint parameters: "

and finally: 
"Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"e272a9a6fae5b78b\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"wires\":[],\"_wireCount\":0,\"credentials\":{\"user\":\"XXX\",\"password\":\"XXX\"},\"endpoint\":\"CORRECT_IP\",\"securityPolicy\":\"Basic256Sha256\",\"securityMode\":\"SignAndEncrypt\",\"login\":true,\"none\":false,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":\"XXX\",\"password\":\"XXX\"}"]"

The issue is that there is nothing wrong with the endpoint, username/password, security policy, as this works fine before a potenial limit(?).  

To fix the error I have to reboot my server. Is this because everytime I redeploy I make a new OPC Node at the server and I exceed the number of OPC nodes connection at the server?  I was able to use UA expert to connect to the same server while node-red is in this fault. and when checking the server I have only witnessed 4 OPC nodes being connected

mikakaraila, 2023-10-03:
Most probably you have exceeded sessions that server can provide.
You can refactor your flow to use one client node and inject msg with action into the node then it will use one session.

Theodorska1, 2023-10-03:
Thank you for your response. 

I have to use one opc client to subscribe and one to write as i making an RTU between OPC UA and MQTT sparkplug B.
Currently it is just me connecting to the OPC server, current session count and subscription count is 1 at the moment, checking with OPC UA Expert. As I haven't seen connected clients above 4, i doubt that this may be the issue. Or is sessions something else?

It is struggeling to reconnect. Especially when I am using Node-Red 3.1, it reconnects easier with node-red 3.0.2

mikakaraila, 2023-10-03:
Hmm, then it is something in your environment if it depends on node-red version.

How you use reconnect? Remember to disconnect() first otherwise it could cause resource lean on server side.

Theodorska1, 2023-10-03:
Thanks again for your swift reply!

I have reverted to version 3.0.2 as this was more stable for my application, I tried the same flow in 3.1, but it did not reconnect for some reason on the specific unit i am using it (it did work on my test bench for some reason). 
so everytime i redeploy I should push a msg.topic =  "disconnect" followed by a msg.topic = 'connect' for the subscription and browse node?
And should i push msg.topic = 'clearitems' for the read multiple nodes?

Is there a documentation for the types of topics I can push into nodes?

f.x. "msg.topic = 'currentSession';
msg.payload = { "action": "deletesubscription" }" I only found in your examples

mikakaraila, 2023-10-03:
There are msg.topic that contains nodeId typically or to content handling.

Then msg.payload.action is for the functionality. I haven´t documented everything, sorry.
But most of actions are same as in the UI. Some are still undocumented. I should do that...

Theodorska1, 2023-10-04:
I am still having the issue when using a specific device.
When using UA Expert I am getting the error BadCertificateHostNameInvalid, which I can ignore and still connect. 
Are there any workaround to ignore this error in your nodes?

Theodorska1, 2023-10-04:
> I am still having the issue when using a specific device. When using UA Expert I am getting the error BadCertificateHostNameInvalid, which I can ignore and still connect. Are there any workaround to ignore this error in your nodes?

I am using the exact same server URL as stated in my OPC Server with the IP address. The issue is that it works sometimes, but most of the time i get invalid endpoint

closed 2024-02-01 by mikakaraila