mikakaraila/node-red-contrib-opcua #614: Dynamically setting the OPCUA endpoint

grcamauer, 2023-10-03:
I am trying to dynamically set the OPCUA endpoint to call a method through an OPCUA-Methd node:

const output = {
    endpoint : msg.req.body.endpoint,
    objectId : "ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch",
    methodId : "ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch.AdminBatchOPCUAAdvancedControlHandler.ctc",
};

output.inputArguments = [
    { dataType: "String", value: msg.req.body.BatchName},
    { dataType: "String", value: msg.req.body.BatchDescription},
    { dataType: "String", value: msg.req.body.AdminCommandType},
];

output.OpcUaEndpoint = {
    credentials: {},
    endpoint: msg.req.body.endpoint,
    securityPolicy: 'None',
    securityMode: 'None',
    login: false,
    user: undefined,
    password: undefined 
}
return output;

It is not working, I receive the following error from the node: "Method node error on node: ".

How can I set the endpoint?

My full flow:

[
    {
        "id": "c7ed9d758f01d7d0",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "52f3edbc.9091b4",
        "type": "debug",
        "z": "c7ed9d758f01d7d0",
        "name": "Params",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 420,
        "y": 140,
        "wires": []
    },
    {
        "id": "fd93261f.62d3c8",
        "type": "http in",
        "z": "c7ed9d758f01d7d0",
        "name": "Create Batch EP",
        "url": "/createBatchEP",
        "method": "get",
        "upload": false,
        "swaggerDoc": "",
        "x": 140,
        "y": 80,
        "wires": [
            [
                "7541c0a6.b5541",
                "606c75cc.23b8ec"
            ]
        ]
    },
    {
        "id": "783d03af.4301fc",
        "type": "http response",
        "z": "c7ed9d758f01d7d0",
        "name": "Result response",
        "statusCode": "200",
        "headers": {},
        "x": 920,
        "y": 200,
        "wires": []
    },
    {
        "id": "606c75cc.23b8ec",
        "type": "function",
        "z": "c7ed9d758f01d7d0",
        "name": "BuildParams",
        "func": "const output = {\n    endpoint : msg.req.body.endpoint,\n    objectId : \"ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch\",\n    methodId : \"ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch.AdminBatchOPCUAAdvancedControlHandler.ctc\",\n};\n\noutput.inputArguments = [\n    { dataType: \"String\", value: msg.req.body.BatchName},\n    { dataType: \"String\", value: msg.req.body.BatchDescription},\n    { dataType: \"String\", value: msg.req.body.AdminCommandType},\n];\n\noutput.OpcUaEndpoint = {\n    credentials: {},\n    endpoint: msg.req.body.endpoint,\n    securityPolicy: 'None',\n    securityMode: 'None',\n    login: false,\n    user: undefined,\n    password: undefined \n}\n\n\nreturn output;\n\n/*\nmsg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n    credentials: {},\n    endpoint: 'opc.tcp://192.168.178.47:4840',\n    securityPolicy: 'None',\n    securityMode: 'None',\n    login: false,\n    user: undefined,\n    password: undefined \n}\n\nreturn msg;\n*/",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 330,
        "y": 80,
        "wires": [
            [
                "a0d081dc.595a2",
                "52f3edbc.9091b4"
            ]
        ]
    },
    {
        "id": "4638ecbd.8cf4e4",
        "type": "function",
        "z": "c7ed9d758f01d7d0",
        "name": "BuildReturnMsg",
        "func": "context.data = context.data || {};\n\nswitch (msg.topic) {\n    case \"reqres\":\n        context.data.reqres = msg;\n        msg = null;\n        break;\n    case \"opcua\":\n        context.data.opcua = msg;\n        msg = null;\n        break;\n    default:\n        msg = null;\n    \tbreak;\n}\n\nif(context.data.reqres != null && context.data.opcua != null ) {\n    var newMsg = context.data.opcua;\n\tnewMsg.req = context.data.reqres.req;\n    newMsg.res = context.data.reqres.res;\n   /* msg.payload.statusCode = context.data.opcua.result.statusCode.value;*/\n    /*newMsg.payload = context.data.opcua.result.outputArguments[0].value;*/\n    newMsg.payload = {msg: context.data.opcua.result.outputArguments[0].value,statusCode:context.data.opcua.result.statusCode.value};\n    context.data=null;\n\treturn newMsg;\n} else return msg;\n",
        "outputs": 1,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 700,
        "y": 200,
        "wires": [
            [
                "4c2da975.99adc8",
                "783d03af.4301fc"
            ]
        ]
    },
    {
        "id": "7541c0a6.b5541",
        "type": "change",
        "z": "c7ed9d758f01d7d0",
        "name": "Pass on req and res",
        "rules": [
            {
                "t": "set",
                "p": "topic",
                "pt": "msg",
                "to": "reqres",
                "tot": "str"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 360,
        "y": 200,
        "wires": [
            [
                "831eff61.4c76b",
                "4638ecbd.8cf4e4"
            ]
        ]
    },
    {
        "id": "831eff61.4c76b",
        "type": "debug",
        "z": "c7ed9d758f01d7d0",
        "name": "Method Vars",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 570,
        "y": 260,
        "wires": []
    },
    {
        "id": "4c2da975.99adc8",
        "type": "debug",
        "z": "c7ed9d758f01d7d0",
        "name": "WebBundle",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 910,
        "y": 260,
        "wires": []
    },
    {
        "id": "f3500478.d31bb8",
        "type": "debug",
        "z": "c7ed9d758f01d7d0",
        "name": "Result",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 890,
        "y": 80,
        "wires": []
    },
    {
        "id": "53a549da.5dd158",
        "type": "change",
        "z": "c7ed9d758f01d7d0",
        "name": "",
        "rules": [
            {
                "t": "set",
                "p": "topic",
                "pt": "msg",
                "to": "opcua",
                "tot": "str"
            }
        ],
        "action": "",
        "property": "",
        "from": "",
        "to": "",
        "reg": false,
        "x": 690,
        "y": 140,
        "wires": [
            [
                "4638ecbd.8cf4e4"
            ]
        ]
    },
    {
        "id": "a0d081dc.595a2",
        "type": "OpcUa-Method",
        "z": "c7ed9d758f01d7d0",
        "endpoint": "372390d2881694df",
        "objectId": "ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch",
        "methodId": "ns=2;s=AdvancedMethods.OPCUA_Method_AdminBatch.AdminBatchOPCUAAdvancedControlHandler.ctc",
        "name": "CreateBatch",
        "inputArguments": [],
        "arg0name": "BatchName",
        "arg0type": "String",
        "arg0typeid": "",
        "arg0value": "",
        "arg1name": "BatchDescription",
        "arg1type": "String",
        "arg1typeid": "",
        "arg1value": "",
        "arg2name": "AdminCommandType",
        "arg2type": "String",
        "arg2typeid": "",
        "arg2value": "",
        "out0name": "ReturnCode",
        "out0type": "Int32",
        "out0typeid": "",
        "out0value": "",
        "x": 570,
        "y": 80,
        "wires": [
            [
                "f3500478.d31bb8",
                "53a549da.5dd158"
            ]
        ]
    },
    {
        "id": "372390d2881694df",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://0.0.0.0:4840/",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]

Thanks,

Guillermo

mikakaraila, 2023-10-04:
Hmm, server cannot change it´s own endpoint.
But here is my example how client node can be dynamically change connection parameters:
[RECONNECT.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12799583/RECONNECT.txt)

mikakaraila, 2025-06-27:
I expect this can be closed