mikakaraila/node-red-contrib-opcua #618: Change localeId for OpcUa-Client to "de-DE"

CA1458, 2023-10-10:
Hello,

I am able to get alarms&events via opc-ua from a Siemens 840d sl machine with the nodes OpcUa-Event and OpcUa-Client.
I get the alarms in the language "en-EN".

In UAExpert I can change the localeId from the client to "de-DE" and then I get the alarms in German.
How can I fix this in nodeRED?

Thank you and best regards

mikakaraila, 2023-10-10:
Not implemented nor supported at the moment. This will need code changes...

CA1458, 2023-10-10:
> Not implemented nor supported at the moment. This will need code changes...

Hmm ok.
Can I change this on my own in the code? If Yes, where?

mikakaraila, 2023-10-10:
I expect this could be place to add check and select wanted locale in case of field that is type of LocalizedText:
[image]

CA1458, 2023-10-10:
> I expect this could be place to add check and select wanted locale in case of field that is type of LocalizedText: [image]

Hello @mikakaraila,
thanks for the image.
Unfortunately, I don't know what new code I have to add here...

mikakaraila, 2023-10-10:
So it´s not so obvious to make changes as I expected... if I have time this or next week I will take a look.
Just one if + some code and settings from somewhere... cannot promise this 100%.

CA1458, 2023-10-10:
> So it´s not so obvious to make changes as I expected... if I have time this or next week I will take a look.
> Just one if + some code and settings from somewhere... cannot promise this 100%.

ok thank you!

mikakaraila, 2023-10-10:
UaExpert setting:
[image]

Corresponding in node-red:
https://groups.google.com/g/node-red/c/tg50hmJ0b8c

Or use directly OS locale inside my code...

CA1458, 2023-10-10:
> UaExpert setting: [image]
> 
> Corresponding in node-red: https://groups.google.com/g/node-red/c/tg50hmJ0b8c
> 
> Or use directly OS locale inside my code...

Yes this change I made in UaExpert and then I got the alarms in German.

CA1458, 2023-10-10:
> So it´s not so obvious to make changes as I expected... if I have time this or next week I will take a look. Just one if + some code and settings from somewhere... cannot promise this 100%.

@mikakaraila maybe you can have a look if you have time
thanks

mikakaraila, 2023-10-12:
Some progress, can you provide me which locales your LocalizedText contains?

12 Oct 13:41:03 - [debug] [OpcUa-Client:A & C] Event Field: Message [{"dataType":"1","arrayType":"2","value":"3"},"LocalizedText","Scalar",{"locale":"4","text":"5"},"","Level exceeded"]
OS Locale: en-FI
Locale: 
Message: Level exceeded

I will use default text but if OS Locale matches with containing LocalizedText message locale then it will be used.

CA1458, 2023-10-12:
> Some progress, can you provide me which locales your LocalizedText contains?
>
> 12 Oct 13:41:03 - [debug] [OpcUa-Client:A & C] Event Field: Message [...]

how can I get this info in nodeRED?

CA1458, 2023-10-12:
This is what I get in NodeRED

[image]

mikakaraila, 2023-10-12:
UaExpert and how many message LocalizedText it will contain?

CA1458, 2023-10-13:
> UaExpert and how many message LocalizedText it will contain?

UaExpert:
[image]

[image]

mikakaraila, 2023-10-13:
Hmm, it shows only the current locale message. I made change and published new version, test with it.
But if you look same structure in Data view you should see that message LocalizedText should be array.
My code will try to select current OS locale text from the array.

CA1458, 2023-10-13:
> Hmm, it shows only the current locale message. I made change and published new version, test with it. But if you look same structure in Data view you should see that message LocalizedText should be array. My code will try to select current OS locale text from the array.

I testet the new version v0.2.315, but the result is the same...

It would be great, If I could type in the desired localeId language in NodeRed opc-ua-client like in UaExpert...

mikakaraila, 2023-10-13:
What comes to console? I added there output to see what happens as I cannot test this with my servers.

CA1458, 2023-10-13:
I`m running nodeRED on Windows
Where can I find the console for NodeRED?

mikakaraila, 2023-10-13:
If you start it from the command line: node-red FLOWFILE.json
Then it should be there...

CA1458, 2023-10-13:
There I get a error message for the flow credentials, that they are encryped.

I have also a test NodeRED running on Ubunto. The System language is German.
Here are the screens:

[image]
[image]
[image]

CA1458, 2023-10-17:
Hi @mikakaraila,

on Windwos I have the same output in console.
Do you have any idea?

[image]

mikakaraila, 2023-10-19:
Look alarm object an message variable, does it contain array of LocalizedText?
[image]

CA1458, 2023-10-19:
Hello @mikakaraila,
this is what I get in UaExpert, if I choose the SINUMERIK in the alarm event:

[image]

CA1458, 2023-10-19:
I have no massage or alarm object. 
Sinumerik has a own alarm event

mikakaraila, 2023-10-19:
Deep dive => open/browse what is inside address space under Sinumerik?

mikakaraila, 2023-10-19:
It seems that we are missing ActivateSession with localeIds.
So there is no API for client to request messages with another locale.

CA1458, 2023-10-19:
@mikakaraila I have opened the Sinumerik and activate all

[image]

mikakaraila, 2023-10-19:
Look address space tree on the left, but anyway there is no support in SDK for this feature. So I cannot implement it.

robotsox, 2024-08-12:
Is here any Update? i need this feature too to change the language of the messages