mikakaraila/node-red-contrib-opcua #621: Multiple OPC UA clients using same node

TobiasKorn, 2023-10-13:
I plan a node red concept, which does not require a programmer to configure OPC UA clients. Configuration of OPC UA endpoints, items, actions, etc. should be set in the browser view, be stored in node red context and send in a msg-object to the OPC client.

I know that it is possible to change the OPC endpoint dynamically, see
https://github.com/mikakaraila/node-red-contrib-opcua/issues/59

But that means, that only one OPC UA session is active. The session which was active before will be closed. Thats why I wish that it would be possible that one OPC Client node could handle multiple clients at the same time. Depending on the msg-object one of the active client sessions could be selected for each request.

What do you think of these concept? Do you think it could be implemented?

mikakaraila, 2023-10-13:
No as OPC UA client needs session to be open - close - disconnect. Session is client specific, it cannot handle multiple servers.

TobiasKorn, 2023-10-13:
Thanks for your quick reply. I know it's not possible now. But do you think it could be an option in the future, that one node can handle multiple sessions / clients? Just from the technical point of view it should be possible, because now it is possible that N client nodes handle N clients. So why shouldn't it be possible to have the same functionality in one node instead.

As mentioned the benefit would be to save programming work, beeing able to expand OPC network dynamically just by configuring data.