mikakaraila/node-red-contrib-opcua #623: OPC UA Custom events support

caccia86, 2023-10-16:
Hi, I need to obtain some specific data from a client opc ua node.
I tried many different ways to do that but with no results.
As the image below, I need the informations about the highlighted event:
[image]
[image]

Here is the flow that I set:
[image]

It shows only those information from debug node.
For example If I try to set the Event Type -> ns=3;i=1038 I see some data retrieved:
[image]
[image]
[image]

I don't know what I'm wrong.
Thanks in advance for whatever support.

caccia86, 2023-10-18:
I add a part of logs picked from function pm2 logs:

[pm2_logs.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/12995756/pm2_logs.txt)

mikakaraila, 2023-10-18:
Server is overloaded, first line: BadTooManyOperations

caccia86, 2023-10-19:
@mikakaraila I imagined that was the problem and for example is it exists a way  to split the request or set a specific field to obtain in output?

caccia86, 2023-10-26:
We found a workaround: in the 102-opcuaclient.js file the async function subscribe_events_async(msg) function is invoked to subscribe events.

When it's used for a standard event, e.g. BaseEvent (all), the eventTypeId variable assume the value "i=2041".

From the log we see that the row below

var fields = await opcua.extractConditionFields(node.session, eventTypeId);

It fills the fields on this way:

[debug] [OpcUa-Client:Prosys events] EventFields: EventId,EventType,SourceNode,SourceName,Time,ReceiveTime,LocalTime,Message,Severity,ConditionId

When we used custom events, the eventTypeId becomes "ns=5;s=GoodShots" or "ns=5;i=6040".
In this case extractConditionFields can't resolve custom fields and in the log we see only one field get back:

[debug] [OpcUa-Client:Prosys events] EventFields: ConditionId

We add a workaround in the 102-opcuaclient.js file:

```
// If field "ConditionClassId" is part of the list, it is or inherits from ConditionType
      if (!fields.includes("ConditionClassId")) {
//        fields.splice(fields.indexOf("ConditionId"), 1); // remove field ConditionId ??? Needed for Acknowledge
      }

      if (msg.hasOwnProperty("customEventFields")){
              verbose_log("MSG has customEventFields");
        if(Array.isArray(msg.customEventFields)){
          verbose_log("customEventFields: " + msg.customEventFields);
          fields = fields.concat(msg.customEventFields);
        } else { verbose_log("!!!customEventFields NOT isArray "); }
      } else { verbose_log("!!!MSG NOT Has customEventFields ");}

      msg.eventFilter = opcua.constructEventFilter(fields, opcua.ofType(eventTypeId));
      msg.eventFields = fields;
      verbose_log("EventFields: " + msg.eventFields);
```

Now we can exploit the BaseEvent to obtain also the custom fields:

[image]

[image]

[debug] [OpcUa-Client:Prosys events] EventFields: EventId,EventType,SourceNode,SourceName,Time,ReceiveTime,LocalTime,Message,Severity,ConditionId,5:GoodShots,3:CycleTime

Fixed output:

[image]

mikakaraila, 2023-10-27:
Can you make PR or send me modified files?
I have been sick for a while and seems to take more days/weeks/months...

mikakaraila, 2023-10-27:
Thanks, I just published new version with this one included.

closed 2023-10-27 by mikakaraila