mikakaraila/node-red-contrib-opcua #625: Once Error forever error

yizenlim, 2023-10-26:
My server is set to allow both Guest authentication and User Authentication, with None security policy.

1) I first send a connect action to the client node -> successfully connected.
2) I send a disconnect action -> successfully disconnected
3) Then I send a new connect with a wrong configuration -> failed to connect. 
4) After that any connects or reconnects with the right configuration will fail until I redeploy the script

[Screen record video ](https://1drv.ms/v/s!Alewio9-RKIng4InAoFyw5-4Ax3X6A?e=Dfs5t3)

mikakaraila, 2023-10-27:
Very special case, I expect I cannot fix this now.
I have been sick for a while and seems to take more days/weeks/months...

yizenlim, 2023-10-27:
> Very special case, I expect I cannot fix this now. I have been sick for a while and seems to take more days/weeks/months...

Hi @mikakaraila I solved the case looking through the code, it was due to userIdentity is only registered once and never get updated in every connect_opcua_client, so updating userIdentity everytime we run a connect will solve the case. 

[image]

This is where I 've updated it. Maybe we can consider during every processInputMsg so that we always refer to the latest input. 

I'll create a pull request, see if all's good for you. 

Anyways @mikakaraila wish you speedy recovery and get well soon ! and thank you for even creating this package at the first place ! :)

mikakaraila, 2023-10-27:
Thanks, I just published merged new version.

closed 2023-10-27 by mikakaraila