mikakaraila/node-red-contrib-opcua #628: Raspberry pi to PLC S7-1200 connection fail

Diaoko9, 2023-11-06:


Diaoko9, 2023-11-06:
Hi
I have an issue with connecting raspberry pi to S7-1200.
I followed the instructions line to line but at the end opcua client cannot connect.
I have entered the node id, and server address correctly.
Do you have any idea?
I have attached payload message[image]

Thank you

mikakaraila, 2023-11-15:
What parameters you have used inside client node?
None/None Anonymous or some other securityModel/securityPolicy?

Then it could be that you have to trust client certificate on S7-1200 => read Siemens manual.

Diaoko9, 2023-11-19:
Hi
It was None/None Anonymous but I forgot to choose client certificate 1200 on hardware.

Thanks for your attention