mikakaraila/node-red-contrib-opcua #629: Client crash because of uncaught exception

Scaredon, 2023-11-07:
`09:38:17.618Z :client_session_keepalive_manag:130    warning : ClientSessionKeepAliveManager#ping_server   serviceResult = BadLicenseExpired (0x810e0000)
09:38:23.438Z :client_session_keepalive_manag:130    warning : ClientSessionKeepAliveManager#ping_server   serviceResult = BadLicenseExpired (0x810e0000)
09:38:27.413Z :client_secure_channel_layer   :236   requestData not found for requestId =  1830 try with  1831
7 Nov 10:38:27 - [red] Uncaught Exception:
7 Nov 10:38:27 - [error] TypeError: Cannot read properties of undefined (reading 'callback')
    at MessageBuilder.<anonymous> (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:243:42)
    at MessageBuilder.emit (node:events:514:28)
    at MessageBuilder._report_error (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\dist\source\message_builder_base.js:128:14)
    at MessageBuilder._feed_messageChunk (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\dist\source\message_builder_base.js:193:22)
    at PacketAssembler.<anonymous> (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\dist\source\message_builder_base.js:64:66)
    at PacketAssembler.emit (node:events:514:28)
    at PacketAssembler.feed (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-packet-assembler\dist\packet_assembler.js:75:18)
    at MessageBuilder.feed (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-transport\dist\source\message_builder_base.js:94:35)
    at ClientSecureChannelLayer._on_receive_message_chunk (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:950:29)
    at ClientTCP_transport.<anonymous> (C:\Users\XXX\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua-secure-channel\dist\source\client\client_secure_channel_layer.js:795:18)`

mikakaraila, 2023-11-15:
What happened? Can you repeat crash?
Seems to crash inside other package at the network level...

Scaredon, 2023-11-15:
Hi @mikakaraila, thanks for your reply.
Yes, this crash is repeatable.
the opcua server running on a demo licence. After a certain time it kinda stops (don't now what they exactly do there) but it seems that it closes the session with status code "BadLicenseExpired". And this leads to that crash.

mikakaraila, 2023-11-15:
What your flow is doing? I mean reading with interval or subscribed items or writing a value?
I would like to see what happens just before it crashes...

Scaredon, 2023-11-15:
I have to create a small example to narrow it down a bit more. Currently there are several clients with different settings in the flows. I don't know which one fails at first.

The server (closed SDK) is currently part of a test setup to evaluate some ideas. As I don't have purchased a licence for that SDK, it stops working after a certain time. (It looks like as the server closes the session to the client with Bad - ErrorCode.)
And then node-red crashes.

is it helpful to create a small setup?
it would be also possible to send you the server (windows only currently)