mikakaraila/node-red-contrib-opcua #631: Wrong Endpoint Parameters

SPKChaosPhoenix, 2023-11-22:
When trying to connect to a OCPUA Server with a node red Client node i get the following error message:
`[31mClient node error on: SPS error: ["Wrong endpoint parameters: {\"id\":\"78292d59.0f1334\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://192.168.11.20:48030\",\"securityPolicy\":\"Basic256Sha256\",\"securityMode\":\"Sign\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"][39m`

I copied all Settings from UAexpert, and the Trusted the Certificate in the opcua server settings.

I can connect to the OPCUA-Server using UAexpert, but have to "Ignore" the following messages to do so:

[image]
and
[image]

mikakaraila, 2023-11-22:
Fix certificate, node-opcua session does not give options to skip verification.

sniicker, 2024-01-04:
Got the same error after updating from Version 0.2.238 to 0.2.320

Here is the cmd log:
> 4 Jan 11:50:14 - [error] [OpcUa-Client:a5d600c0.43c2d] Invalid endpoint
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : can't work without OPC UA client
  defaultSecureTokenLifetime.... 200000
  securityMode.................. None
  securityPolicy................ http://opcfoundation.org/UA/SecurityPolicy#None
  certificate fingerprint....... f9f9bccad3c0a25153d91ee47d1546a36338afdd
  server certificate fingerprint
  keepSessionAlive.............. false
  bytesRead..................... 163
  bytesWritten.................. 210
  transactionsPerformed......... 1
  timedOutRequestCount.......... 0
  connectionStrategy.
        .maxRetry............... 10512000
        .initialDelay........... 5000
        .maxDelay............... 30000
        .randomisationFactor.... 0
  keepSessionAlive.............. false
  applicationName............... NodeOPCUA-Client
  applicationUri................ urn:IE-N01:NodeOPCUA-Client
  clientName.................... ClientSession
  reconnectOnFailure............ true
  isReconnecting................ true
  (internal state).............. disconnected
  sessions count................ 0
  requestedSessionTimeout....... 300000
  endpointUrl................... opc.tcp://localhost:54840/Pirlo-OPC-UA/Compact
  serverUri..................... undefined
 client undefined
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : Creating OPCUA CLIENT
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : connect_opcua_client
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : userIdentity is ANONYMOUS
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] :         using securityPolicy: ["None"]
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : Case C: With Sign you cannot use SecurityPolicy None!!
4 Jan 11:50:14 - [warn] [OpcUa-Client:a5d600c0.43c2d] : Invalid endpoint parameters:
4 Jan 11:50:14 - [error] [OpcUa-Client:a5d600c0.43c2d] Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"672de13ef73fb94f\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://localhost:54840/Pirlo-OPC-UA/Compact\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"]
4 Jan 11:50:14 - [error] [OpcUa-Client:a5d600c0.43c2d] Invalid endpoint

It worked the past 3 years. Always used user & password. Now can't even connect anonymous.
With UAExpert no Problem.

Have a idea?
thx

sniicker, 2024-04-18:
Trying since days now....
Do not get i working again. Even did a full re-install on my system, deleted all certificate folders, but nothing works..
Even tried to make my own certificate. 
It doesn't matter what i set up in the client config, it always says: Invalid endpoint parameters

`Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"0d8b6a055532c79f\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{\"user\":\"****\",\"password\":\"*******\"},\"endpoint\":\"opc.tcp://localhost:54840/Pirlo-OPC-UA/Compact\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":\"*****\",\"password\":\"*****\"}"]`

It is a little bit  interesting that the error shows the user and pwd even if i set to None and anonymous. 
What can i do..?
As said there is no issue with UA Expert!

Would really appreciate your help
thx mika

mikakaraila, 2024-04-19:
Which environment? UaExpert & node-red on same environment or different machine?
Just to be sure that it is not firewall or ay permission problem...

sniicker, 2024-04-19:
On same Windows 10 machine, same user, all things up to date

sniicker, 2024-05-06:
Any news on that?