mikakaraila/node-red-contrib-opcua #632: Client crash when writing a data type different from the one set in the OpcUa-Item

Pietro395, 2023-11-23:
We noticed that the client crashes when writing a data type different from the one set in the OpcUa-Item.
In particular, setting "String" and injecting a timestamp results in this Uncaught Exception:
```

Session created for node: WRITE TEST
Writing datatype: 12 value: 1700745221747
23 Nov 13:13:41 - [red] Uncaught Exception:
23 Nov 13:13:41 - [error] Error
    at assert (/home/intac/.node-red/node_modules/node-opcua-assert/dist/index.js:14:21)
    at coerceVariantType (/home/intac/.node-red/node_modules/node-opcua-variant/dist/variant.js:660:44)
    at constructHook (/home/intac/.node-red/node_modules/node-opcua-variant/dist/variant.js:394:25)
    at new Variant (/home/intac/.node-red/node_modules/node-opcua-variant/dist/variant.js:84:26)
    at write_action_input (/home/intac/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1826:24)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (/home/intac/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:803:11)
    at /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:210:26
    at Object.trigger (/usr/lib/node_modules/node-red/node_modules/@node-red/util/lib/hooks.js:166:13)
    at OpcUaClientNode.Node._emitInput (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:202:11)
    at OpcUaClientNode.Node.emit (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:186:25)
```

[flows_crash.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/13450324/flows_crash.json)

mikakaraila, 2023-11-24:
Fix your flow first, I will look this later. No time at the moment.

Pietro395, 2023-11-24:
> Fix your flow first, I will look this later. No time at the moment.

Yes we fixed it, we just wanted to point this out :)