mikakaraila/node-red-contrib-opcua #637: Missing values on subscription

vdwpsmt, 2023-12-06:
Hi, 

I have an opc-ua client with a subscription interval of 200 ms on a variable of type "ExtensionObject"
This works fine for most of the time, but in  ca. 1 out of 100 data changes (randomly) , the client doesn't receive the data.
The data is a simple counter, added by 1 every 300 ms, so verification is pretty straightforward.  
I used another opc-ua client to compare and on this other client I do receive all the data.

How can I debug this problem?  I checked the node-red logging, nothing found concerning the opc-ua client.  Is there anything else I can check?  Can I add extra logging to the opc-ua node to verify?

thanks

mikakaraila, 2023-12-07:
Your can first turn on from the settings.js (node-red) debug level on.
Then another low level tracing can be done setting environment variable before you start node-red:
set NODEOPCUADEBUG="CLIENT{TRACE-REQUEST-RESPONSE}"

BUT look first what subscription parameters you are using, it should come with verbose logging level:
verbose_log("Publishing interval " + stringify(parameters));

alexTingvalla, 2024-06-03:
Hello,

I'm experiencing the same problem. In my Node-RED flow, an OPC UA client with a subscribe action receives a signal from an OPC server. Every 5 seconds or so, the signal changes to true for about 100 ms and then becomes false again. About 5% of these value changes are being missed in Node-RED, even though the interval is set to 50 ms.

Is there a known lower limit for the duration of a signal to guarantee that the OPC subscriber in Node-RED doesn't miss it? 
Is there a clever way to solve this problem? Maybe some kind of handshake between the subscriber and the server? 
Would a faster CPU on my device make any difference?

The only solution I can come up with is to edit the PLC program. In that case, the PLC would set a variable when the sensor signal becomes true, send the variable to Node-RED, and not reset the variable until the PLC receives a confirmation signal from Node-RED confirming that the variable has been read successfully.

I set the NODEOPCUADEBUG=.... enviroment varible and took a screenshot of the log. 
To me it looks like there is approx. 300 ms between the request and the response.

[image]