mikakaraila/node-red-contrib-opcua #638: Invalid Channel BadConnectionClosed

keyadelta, 2023-12-09:
Hello, i've been working on monitoring system using this opc ua node. The case if whenever the machine turned off, the opc ua client will show the error "invalid Channel BadConnectionClosed". I've tried using an inject node with message "reconnect" with interval  every morning at 6 am but the reconnect node made another error where the opc ua client won't close the session and renew it. Hereby i attached some of the logs when the error occurs and does anybody knows how to reconnect the correct way by taking the error message to reconnect by function inject?

8 Dec 13:41:29 - [error] [OpcUa-Client:a54bd6d9bd82477a] Client node error on:  error: [" error at active reading: Invalid Channel BadConnectionClosed"]
8 Dec 13:41:29 - [error] [OpcUa-Client:a54bd6d9bd82477a] Client node error on:  error: [" error at active reading: Invalid Channel BadConnectionClosed"]
8 Dec 13:41:29 - [error] [OpcUa-Client:a54bd6d9bd82477a] Client node error on:  error: [" error at active reading: Invalid Channel BadConnectionClosed"]
8 Dec 13:41:29 - [error] [OpcUa-Client:a54bd6d9bd82477a] Client node error on:  error: [" error at active reading: Invalid Channel BadConnectionClosed"]
06:41:39.806Z :client_session_keepalive_manag:159    warning : ClientSessionKeepAliveManager#ping_server  Invalid Channel BadConnectionClosed

mikakaraila, 2023-12-11:
Look #532 you could try to disconnect first.

keyadelta, 2023-12-13:
[image]

I've tried using disconnect using manual inject, but i wanted it to automatically reconnect when a specific error occurs, so i used this function to inject "reconnect" when "Error: Invalid Channel BadConnectionClosed"

let errorOccurredBefore = context.get('errorOccurredBefore') || false;
let error = msg.error;
let status = msg.status;
if (error === "Error: Invalid Channel BadConnectionClosed" && !errorOccurredBefore) {
  msg.topic = "Reconnect";
  msg.action = "reconnect";
  msg.opcUAEndpoint = {
    credentials: {},
    endpoint: "opc.tcp://192.168.48.14:4840",
    securityPolicy: "None",
    securityMode: "None",
    login: false,
  };
  context.set('errorOccurredBefore', true);
} else {
  context.set('errorOccurredBefore', false);
}
return msg;

thus the function make another error with this log:
13 Dec 15:07:19 - [warn] [OpcUa-Client:opc] opc: can't work without OPC UA NodeId - msg.topic empty

mikakaraila, 2023-12-13:
Another case can be that if you use two computers time difference is more than 60s.

keyadelta, 2023-12-21:
> > Hello @mikakaraila I have the same problem, I've create a status node to monitor the "BadConnectionClosed" message, when this happens I connected an OPC client with the same endpoint with the DISCONNECT action but I have the same error as @keyadelta: `DISCONNECT: can't work without OPC UA NodeId - msg.topic empty"`
> > How should it work? how can I disconnect the client that is stuck with the error "BadConnectionClosed"?
> > Thank you
> 
> I think I solved it with reconnect and a function with this code. I will check the problem in the next few days
> 
> ```
> if (msg.status.text.includes('Badconnect')) {
>     return {action: "reconnect", topic: "reconnect"}
> }
> ```
> 
> ![immagine](https://private-user-images.githubusercontent.com/50447537/290252858-bb6138ba-1f9b-4063-9db5-c6f082e28236.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxMjUwMzIsIm5iZiI6MTcwMzEyNDczMiwicGF0aCI6Ii81MDQ0NzUzNy8yOTAyNTI4NTgtYmI2MTM4YmEtMWY5Yi00MDYzLTlkYjUtYzZmMDgyZTI4MjM2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjIxVDAyMTIxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM0MzQxZGQ0MzZkZmVkODUwOTM1MDAxMDgwZGFkZDg0OGZmZDhiNzU0NGY1N2IxZjkxZTcxNDM4MGZjYTU1OTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2Cu3Z54JMfXse0xnVUPKS8AQrsYEiw8_svm9pvB-Zy8)

is there any update by using this function code ? Does the OPC UA Client still stuck with the badconnectionclosed error?

Pietro395, 2023-12-21:
> is there any update by using this function code ? Does the OPC UA Client still stuck with the badconnectionclosed error?

From our tests seems that this function resolve the prpblem

mikakaraila, 2023-12-21:
Before making PR, I would like to test this with the latest node-opcua.
There has been some updates... you could try to install v2.118.0 and test with it.

Pietro395, 2023-12-21:
> Before making PR, I would like to test this with the latest node-opcua. There has been some updates... you could try to install v2.118.0 and test with it.

We have now installed the version, we are waiting to see if the problem occurs again and we will let you know

Pietro395, 2024-01-11:
> Before making PR, I would like to test this with the latest node-opcua. There has been some updates... you could try to install v2.118.0 and test with it.

We've been using the latest version for a few weeks now, we haven't had the problem, but it could be that it was covered by the reconnect logic I was explaining in previous comments

eduardo-leite-oliveira, 2024-05-23:
> I think I solved it with reconnect and a function with this code. I will check the problem in the next few days
>
> ```
> if (msg.status.text.includes('BadConnect')) {
>     return {action: "reconnect", [...]

Hi Pietro!

I tried to create a function node like you but I received the error: cannot read properties of undefined (reading 'text')

let errorOccurredBefore = context.get('errorOccurredBefore') || false;
let error = msg.error;
let status = msg.status;
let text = msg.status.text;
if (error === "Error: Invalid Channel BadConnectionClosed" && !errorOccurredBefore) {
    msg.topic = "Reconnect";
    msg.action = "reconnect";
    msg.opcUAEndpoint = {
        credentials: {},
        endpoint: "opc.tcp:xxxxxxx",
        securityPolicy: "None",
        securityMode: "None",
        login: true,
        user: "user",
        password: "password"
    };
    context.set('errorOccurredBefore', true);
} else {
    context.set('errorOccurredBefore', false);
}
if (msg.status.text.includes('BadConnect' || 'Connection Break')) {
    return { action: "reconnect", topic: "reconnect" }
}
return msg;

Do you know what is wrong?

mikakaraila, 2025-02-04:
It seems to be that server will response to request with failure and node-opcua will create panic.

Have to check this perhaps with Etienne or with the latest node-opcua