mikakaraila/node-red-contrib-opcua #640: Client Crash on Events

rukinhas, 2023-12-13:
Hello,

I'm trying the example flow for event listening, and only changing the OPC Server address.
When I inject , the client crashes with the following error:
` 13 Dec 18:23:19 - [red] Uncaught Exception:
 13 Dec 18:23:19 - [error] Error:  cannot find conditionRefreshId
     at /data/node_modules/node-opcua-client/source/alarms_and_conditions/client_tools.ts:58:35
     at /data/node_modules/node-opcua-client/source/private/client_session_impl.ts:1459:13
     at /data/node_modules/node-opcua-client/source/private/client_session_impl.ts:1554:13
     at ClientSecureChannelLayer.<anonymous> (/data/node_modules/node-opcua-client/source/private/client_session_impl.ts:1629:20)
     at modified_callback (/data/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1500:31)
     at process_request_callback (/data/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:142:5)
     at ClientSecureChannelLayer._on_message_received (/data/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:963:9)
     at MessageBuilder.<anonymous> (/data/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:481:22)
     at MessageBuilder.emit (node:events:513:28)
     at MessageBuilder._decodeMessageBody (/data/node_modules/node-opcua-secure-channel/source/message_builder.ts:362:26)`

I've tryied the following address, that I got using UAExpert software.
opc.tcp://10.15.50.3:4840/comoneo/server/
opc.tcp://10.15.50.3:4840/

I also tryied changing the event type for a custom event ns2;i=1 and ns=2;i=16 (got the ID from UaExpert)

On UAexpert I can get the events without problem.

I'm trying firstly to get all events, But in the future would like get a specific one only.

Any help

mikakaraila, 2023-12-14:
Look other similar issues, you have just wrong parameters...

rukinhas, 2023-12-14:
hello.
Thank you.

I'm using the events example flow, and only changing the server adresss.
Shouldn't I receive some events?

I've checked that the source node in UA Expert is the same on the example flow.