mikakaraila/node-red-contrib-opcua #641: Invalid Channel BadConnectionClosed

TobiasKorn, 2023-12-14:
Hi,

I connect to a OPC server with **Basic256Sha256 Sign&Encrypt** with **username** and **password**. The connection is stable unless the OPC server is shutdown for a longer period (about 16 hours).  Then I regulary receive the error "Invalid Channel BadConnectionClosed" at the client node. Server logs look like this:

_On OPC server boot up (repeating every few seconds):_
06:06:53.210Z :client_secure_channel_layer   :212   requestData not found for requestId =  1 try with  2
06:06:53.211Z :client_base_impl              :1263  _repairConnection already in progress  panic
06:06:53.212Z :reconnection                  :405   session restoration has failed! err = Invalid channel ns=1;i=68580381  => Let's retry
06:06:53.212Z :reconnection                  :408   ... will retry session repair... in 2000 ms
06:06:53.229Z :client_secure_channel_layer   :212   requestData not found for requestId =  1 try with  2
06:06:53.229Z :client_base_impl              :1263  _repairConnection already in progress  panic
06:06:53.229Z :reconnection                  :405   session restoration has failed! err = Invalid channel ns=1;i=1536857049  => Let's retry
06:06:53.229Z :reconnection                  :408   ... will retry session repair... in 2000 ms
06:06:55.227Z :reconnection                  :410   Retrying session repair...
06:06:55.227Z :reconnection                  :405   session restoration has failed! err = Invalid channel ns=1;i=68580381  => Let's retry
06:06:55.228Z :reconnection                  :408   ... will retry session repair... in 2000 ms
06:06:55.243Z :reconnection                  :410   Retrying session repair...
06:06:55.243Z :reconnection                  :405   session restoration has failed! err = Invalid channel ns=1;i=1536857049  => Let's retry
06:06:55.243Z :reconnection                  :408   ... will retry session repair... in 2000 ms
06:06:57.233Z :reconnection                  :410   Retrying session repair...
06:06:57.234Z :reconnection                  :405   session restoration has failed! err = Invalid channel ns=1;i=68580381  => Let's retry
06:06:57.234Z :reconnection                  :408   ... will retry session repair... in 2000 ms
06:06:57.249Z :reconnection                  :410   Retrying session repair...
06:06:57.249Z :reconnection                  :405   session restoration has failed! err = Invalid channel ns=1;i=1536857049  => Let's retry
06:06:57.249Z :reconnection                  :408   ... will retry session repair... in 2000 ms

_On inject:_
14 Dec 07:08:18 - [error] [OpcUa-Client:f9823d9231f003f0] Client node error on:  error: ["Session is not active!"]
14 Dec 07:08:18 - [error] [OpcUa-Client:f9823d9231f003f0] Client node error on:  error: ["Session is not active!"]
14 Dec 07:08:18 - [error] [OpcUa-Client:5d494271edd1e21e] Client node error on:  error: [" **error at active reading: Invalid Channel BadConnectionClosed**"]

By the way: I never request nodes from namespace 1 only 3. So i wonder why it is tries reading those.
Thank you for your support

mikakaraila, 2023-12-14:
I expect the node id is something internal inside node-opcua as it will handle automatic reconnecting. Is server providing method to resendData to retrieve subscription data?

It could be related to it. If you want to solve this then you should ask official support from Sterfive (node-opcua).

TobiasKorn, 2023-12-14:
The node does indead try to handle reconnecting automatically. I can see all the reconnect attempts in the logfile. Often the node is  reconnecing successfully, but regulary after longer shutdown of OPC server not. However we use also the same OPC server with same security settings somewhere else, which does not has this problem.

The logs above show the behaviour when the client tries to reconnect, but somehow fails. Then it stops trying reconnecting, instead it wants to repair the session and remains in this state.

After redeploy in node-red the connection is coming directly back again, because it forces the client to kill all sessions and create new ones i guess.

mikakaraila, 2024-02-01:
Please re-test with the latest version.

closed 2024-02-01 by mikakaraila