mikakaraila/node-red-contrib-opcua #642: Disconnecting OPCUA Client causes Node-Red crash

scrow66, 2023-12-18:
Hallo, when send msg.action = "disconnect" to an OPC UA Client Node the following error crashes Node-Red

```
Uncaught Exception:
TypeError: Cannot read properties of undefined (reading 'endpoint')
    at set_node_status_to (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:450:36)
    at /data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1215:11
    at /data/node_modules/node-opcua-client/dist/private/client_base_impl.js:985:17
    at /data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:496:17
    at modified_callback (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1006:17)
    at process_request_callback (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:68:5)
    at ClientSecureChannelLayer._cancel_pending_transactions (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:623:21)
    at ClientSecureChannelLayer._on_transport_closed (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:636:14)
    at ClientTCP_transport.<anonymous> (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:784:51)
    at ClientTCP_transport.emit (node:events:513:28)
```

[image]

scrow66, 2023-12-18:
Thanks for the quick reply, I used the provided [connection example](https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/examples/OPCUA-CONNECT.json), just replaced the OPC UA item and the endpoint. Here is my flow

[opcua-example.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/13706148/opcua-example.json)

mikakaraila, 2023-12-20:
[opcua-disconnect.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/13724261/opcua-disconnect.json)

Cannot repeat problem with Prosys UA Simulation server.
I expect one reason could be that your counter variable is defined as Boolean not Double.

RudolfTheRedNode, 2024-01-31:
Hi Mika, 

I also experience this problem. When disconnecting I got this error shown below:
When I inject disconnect using function node:

========Function node program below=========
msg.topic="Disconnect";
msg.action="disconnect";
var opcIP=global.get('OPCEndpoint');
var opcendpoint = "opc.tcp://" + opcIP + ":4840/";

msg.OpcUaEndpoint = {
    credentials:{
        user: 'admin',
        password: 'admin'
    },
    endpoint: opcendpoint,
    securityPolicy: 'None',
    securityMode: 'None',
    login: true,
    user: 'admin',
    password: 'admin',
}
===================================

======Error from debug window============
 OPC Client: can't work without OPC UA client null client  name..................... ClientSession1
 sessionId................ ns=1;g=6B59C1D4-C423-6E91-D26D-A6CA8358B56B
 authenticationToken...... ns=1;g=62DCEB51-67CA-BD04-CD3C-8358F09F69D6
 timeout.................. 300000ms[32m timeout in 298.635 seconds[39m
 serverNonce.............. 885c0ee09d8f08f81079f9063842114bf9144ac1f3f3919d1679eb12dd8e57ec
 serverCertificate........ 308203853082026da003 ... dd98979625dcae7811e3
 serverSignature.......... {[36m /*SignatureData*/[39m
[33m algorithm                    [39m [36m/*  String                                    */[39m: [34mnull[39m
[33m signature                    [39m [36m/*  ByteString                                */[39m: [34mnull[39m
};
 lastRequestSentTime...... 2024-01-31T06:15:50.819Z  (1365)
 lastResponseReceivedTime. 2024-01-31T06:15:50.818Z (1366)
 isReconnecting........... false
 isValidChannel........... true has been closed  true
 channelId............

====================================================================

neben, 2024-02-28:
This also happens with the ProSys UA Simulation server. Same as @scrow66 I used the connection example from this repo, which doesn't have an endpoint set in the `msg`. Which makes sense, as this is already set in the client node. Running the `disconnect` action triggers the above mentioned uncaught exception in node red, thus terminating the process.

```
28 Feb 20:08:50 - [red] Uncaught Exception:
28 Feb 20:08:50 - [error] TypeError: Cannot read properties of undefined (reading 'endpoint')
    at set_node_status_to (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:449:36)
    at /data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1219:11
    at /data/node_modules/node-opcua-client/dist/private/client_base_impl.js:985:17
    at /data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:496:17
    at modified_callback (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1006:17)
    at process_request_callback (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:68:5)
    at ClientSecureChannelLayer._cancel_pending_transactions (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:623:21)
    at ClientSecureChannelLayer._on_transport_closed (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:636:14)
    at ClientTCP_transport.<anonymous> (/data/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:784:51)
    at ClientTCP_transport.emit (node:events:513:28)
```

The problem is in https://github.com/mikakaraila/node-red-contrib-opcua/blob/935ecba0db8ded43b86953a72757b418bca08da4/opcua/102-opcuaclient.js#L449 (among other places), where `opcuaEndpoint` is dereferenced even though it might not be defined. In other places in the code there is a guard that checks it exists, but that's missing here (and in other places). So not sure why this is undefined in the first place (why does it rely on msg having the endpoint set and not the client?), but in any case this should not cause an unhandled exception. So either these places should have checks, or use the [optional chaining operator](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Operators/Optional_chaining).

TheeeEnding, 2024-04-29:
Hi, I've seen the same behavior for a few weeks now with v0.2.323 and v0.2.325.
At first this error was shown:

`2024-04-29 07:54:54.531 | 05:54:54.530Z :client_secure_channel_layer   :1040  ERROR !!! callback has thrown en error  TypeError: Cannot read properties of undefined (reading 'endpoint')`

And then the same uncaught exception as mentioned before by others:
`<html>
<body>

2024-04-29 07:54:54.538 | 29 Apr 05:54:54 - [red] Uncaught Exception: |  
-- | -- | --
  |   | 2024-04-29 07:54:54.538 | 29 Apr 05:54:54 - [error] TypeError: Cannot read properties of undefined (reading 'endpoint') |  
  |   | 2024-04-29 07:54:54.538 | at set_node_status_to (/usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:450:36) |  
  |   | 2024-04-29 07:54:54.538 | at /usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1220:11 |  
  |   | 2024-04-29 07:54:54.538 | at /usr/src/node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:1002:17 |  
  |   | 2024-04-29 07:54:54.538 | at /usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:496:17 |  
  |   | 2024-04-29 07:54:54.538 | at modified_callback (/usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1041:17) |  
  |   | 2024-04-29 07:54:54.538 | at process_request_callback (/usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:69:5) |  
  |   | 2024-04-29 07:54:54.538 | at ClientSecureChannelLayer._cancel_pending_transactions (/usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:623:21) |  
  |   | 2024-04-29 07:54:54.538 | at ClientSecureChannelLayer._on_transport_closed (/usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:639:14) |  
  |   | 2024-04-29 07:54:54.538 | at ClientTCP_transport.<anonymous> (/usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:806:51) |  
  |   | 2024-04-29 07:54:54.538 | at ClientTCP_transport.emit (node:events:517:28) |  
  |   | 2024-04-29 07:54:54.538 | at ClientTCP_transport._emitClose (/usr/src/node-red/node_modules/node-opcua-transport/dist/source/tcp_transport.js:394:18) |  
  |   | 2024-04-29 07:54:54.538 | at ClientTCP_transport._on_socket_close (/usr/src/node-red/node_modules/node-opcua-transport/dist/source/tcp_transport.js:387:14) |  
  |   | 2024-04-29 07:54:54.538 | at Socket.<anonymous> (/usr/src/node-red/node_modules/node-opcua-transport/dist/source/tcp_transport.js:249:45) |  
  |   | 2024-04-29 07:54:54.538 | at Socket.emit (node:events:517:28) |  
  |   | 2024-04-29 07:54:54.538 | at TCP.<anonymous> (node:net:350:12)

</body>
</html>`

I'm also using the msg.action = "disconnect" with the msg.topic

mikakaraila, 2024-04-30:
Should be OK now

closed 2024-04-30 by mikakaraila