mikakaraila/node-red-contrib-opcua #644: Invalid Endpoint,Check That Server Has Security Policy: ['None']

omerkara1980, 2024-01-04:
Hello,

I want to connect Siemens S7 Plc but I cant connect giving this message.
[image]
This is Debug page.

"[31mClient node error on: OPC UA READ error: ["Wrong endpoint parameters: {\"id\":\"a0e9594fd101860e\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://192.168.0.1:4840\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"][39m"

MBL-DGS, 2024-01-10:
Please give more information about the PLC OPCUA server configuration, since probably there is the problem. Also, first test if you can connect to the server using e.g. UAexpert. That should give you enough debugging tools to find where the problem is.

mikakaraila, 2024-02-01:
Open each client, select endpoint, save and deployed flow, re-test.

closed 2024-02-01 by mikakaraila