mikakaraila/node-red-contrib-opcua #645: Error: no response from Server - Node red OPCUA flow

RajeshShank, 2024-01-09:
### Issue:
Error: no response from server.
When OPCUA node red flow is triggered continuously with an interval of 1 to 3 second. Please review this issue and provide a comment or fix to take it ahead.

[image]

### Server details:
OPCUA server: PROSYS Simulation Server

[image]

### Client node red flow:

[image]

RajeshShank, 2024-01-12:
@mikakaraila - Can you please support on this.

mikakaraila, 2024-01-12:
Did you look examples? Attach your flow here so I can check it.

RajeshShank, 2024-01-12:
Yes, I did. Attached the flow json file for your reference.
[flows.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/13915125/flows.json)
OPCUA server is PROSYS simulation server.

mikakaraila, 2024-01-12:
You have wrong nodeId in the Event node:
[image]

Works:
[image]

RajeshShank, 2024-01-12:
@mikakaraila - Thanks for your support. Can you please share the working flow.json here. I could see that OPC UA Client Node in my flow contains two output ports but there is only one output port in your flow. I wanted to know whether I have to change any settings in OPCUA client node.

mikakaraila, 2024-01-12:
Second port is just for error output.

closed 2024-01-12 by mikakaraila