mikakaraila/node-red-contrib-opcua #647: Custom event causes Node-red Crash; [error] Error: May be subscription is not yet initialized

MBL-DGS, 2024-01-10:
### Issue:
I got a succesfull subscribed node, see below.  
[image]

However, if i extend it by an OPCUA- EVENT like below, 
[image]
then i got a node red crash:

```
nodered   | 09:07:19.405Z :client_session_impl           :96    client is now attempting to recreate a session
nodered   | 09:07:19.462Z :reconnection                  :405   session restoration has failed! err = reconnection cancelled due to session termination ns=1;i=1520392806  => Let's retry
nodered   | 09:07:19.463Z :reconnection                  :416   session restoration should be interrupted because session has been closed forcefully

nodered   | 10 Jan 09:07:20 - [red] Uncaught Exception:
nodered   | 10 Jan 09:07:20 - [error] Error: May be subscription is not yet initialized
nodered   |     at assert (/usr/src/node-red/node_modules/node-opcua-assert/dist/index.js:14:21)
nodered   |     at Object.callConditionRefresh (/usr/src/node-red/node_modules/node-opcua-alarm-condition/dist/call_condition_refresh.js:18:37)
nodered   |     at ClientMonitoredItemImpl.<anonymous> (/usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:2358:17)
nodered   |     at ClientMonitoredItemImpl.emit (node:events:513:28)
nodered   |     at ClientMonitoredItemImpl._after_create (/usr/src/node-red/node_modules/node-opcua-client/dist/private/client_monitored_item_impl.js:259:18)
nodered   |     at /usr/src/node-red/node_modules/node-opcua-client/dist/client_monitored_item_toolbox.js:56:35
nodered   |     at /usr/src/node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:1195:13
nodered   |     at /usr/src/node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:100:17
nodered   |     at ClientSecureChannelLayer.<anonymous> (/usr/src/node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:770:20)
nodered   |     at modified_callback (/usr/src/node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1002:31)

````

### Context:
I'm running node-red on a raspberry pi. 
```
nodered   | [info] Node-RED version: v3.1.0
nodered   | [info] Node.js  version: v16.20.2
nodered   | [info] Linux 5.10.152-rt75-v7l arm LE
```

The events are succesfully created. See the UAexpert view:
[image]

mikakaraila, 2024-01-10:
With custom event you must provide event type. Code does not validate field.

MBL-DGS, 2024-01-10:
But for the event i provide the event type right? Namely, ns=3;i=1805. Which is the same type as the UAexpert is indicating.

MBL-DGS, 2024-01-10:
It did not change anything. I also tried:
 - set the type of the event to BaseEvent instead of a custom type (ns=3;i=1805) 
 - injecting empty string directly
 - injecting empty string once after 1 second
 - changing the source node for the event
 
but all return the same error message

mikakaraila, 2024-01-10:
Look older closed issues... if I remember correctly this is similar (happened earlier).

MBL-DGS, 2024-01-11:
I looked into the closed items, but could not find it. Can you direct me to the referred item?

mikakaraila, 2024-01-11:
Look #623 and #569

mikakaraila, 2024-01-29:
Bug in my code, fixed among other fixes. I will publish new version later today/this week.

closed 2024-02-01 by mikakaraila