mikakaraila/node-red-contrib-opcua #648: node red opc ua endpoint invalid check that server has security policy

matteovi, 2024-01-11:
HI,
I have a peculiar error in the contrib-opcua package.
Assuming that, until yesterday, everything was working normally. Now it seems that the opc ua client can no longer connect to the server.
This is the error log.
Opinions?
`
15:44:36.670Z :client_secure_channel_layer   :1573  xxxxx   >>>>>> request has already been set with a requestHandle 2 0 GetEndpointsRequest
15:44:36.671Z :client_secure_channel_layer   :1579
15:44:36.674Z :client_secure_channel_layer   :1580  Error: Investigate me
 ...                                                    at ClientSecureChannelLayer._internal_perform_transaction (C:\Users\infoconsulting\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1580:22)
 ...                                                    at ClientSecureChannelLayer._performMessageTransaction (C:\Users\infoconsulting\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:1541:14)
 ...                                                    at ClientSecureChannelLayer.performMessageTransaction (C:\Users\infoconsulting\.node-red\node_modules\node-opcua-secure-channel\source\client\client_secure_channel_layer.ts:762:14)
 ...                                                    at OPCUAClientImpl.performMessageTransaction (C:\Users\infoconsulting\.node-red\node_modules\node-opcua-client\source\private\client_base_impl.ts:994:29)
 ...                                                    at OPCUAClientImpl.getEndpoints (C:\Users\infoconsulting\.node-red\node_modules\node-opcua-client\source\private\client_base_impl.ts:1060:14)
 ...                                                    at OPCUAClientImpl.getEndpoints (C:\Users\infoconsulting\.node-red\node_modules\thenify\index.js:65:46)
 ...                                                    at OPCUAClientImpl.getEndpoints (C:\Users\infoconsulting\.node-red\node_modules\node-opcua-client\source\private\client_base_impl.ts:1042:25)
 ...                                                    at OPCUAClientImpl.getEndpoints (C:\Users\infoconsulting\.node-red\node_modules\thenify\index.js:65:46)
 ...                                                    at C:\Users\infoconsulting\.node-red\node_modules\node-opcua-client\source\private\client_base_impl.ts:695:30
 ...                                                    at C:\Users\infoconsulting\.node-red\node_modules\async\dist\async.js:3676:28
11 Jan 16:44:36 - [warn] [OpcUa-Client:PS8] PS8: Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]
11 Jan 16:44:36 - [warn] [OpcUa-Client:PS8] PS8:         using securityPolicy: ["None"]
11 Jan 16:44:36 - [warn] [OpcUa-Client:PS8] PS8: Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined
11 Jan 16:44:36 - [warn] [OpcUa-Client:PS8] PS8: Case C: With Sign you cannot use SecurityPolicy None!!
11 Jan 16:44:36 - [warn] [OpcUa-Client:PS8] PS8: Invalid endpoint parameters:
11 Jan 16:44:36 - [error] [OpcUa-Client:PS8] Client node error on: PS8 error: ["Wrong endpoint parameters: {\"id\":\"01acb1683af3e710\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://10.4.4.12:4840\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"]
11 Jan 16:44:36 - [error] [OpcUa-Client:PS8] Invalid endpoint
15:44:36.715Z :message_chunker               :170   [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 65536 nb chunks 1
15:44:36.718Z :client_secure_channel_layer   :877   { /*ServiceFault*/
 ...                                                 responseHeader                /*  ResponseHeader                            */: {
 ...                                                   timestamp                   /*  DateTime                                  */: 2024-01-11T15:44:36.716Z
 ...                                                   requestHandle               /*  UInt32                                    */: 0               0x0
 ...                                                   serviceResult               /*  StatusCode                                */: BadInternalError (0x80020000)
 ...                                                   serviceDiagnostics          /*  DiagnosticInfo                            */: { /*DiagnosticInfo*/
 ...                                                 namespaceUri                  /*  Int32                                     */: null
 ...                                                 symbolicId                    /*  Int32                                     */: -1
 ...                                                 locale                        /*  Int32                                     */: -1
 ...                                                 localizedText                 /*  Int32                                     */: -1
 ...                                                 additionalInfo                /*  String                                    */: null
 ...                                                 innerStatusCode               /*  StatusCode                                */: Good (0x00000000)
 ...                                                 innerDiagnosticInfo           /*  DiagnosticInfo                            */: null
 ...                                                };
 ...                                                   stringTable                 /*  String                                []  */: [ /* length =1*/
 ...                                                     message size exceeds the negotiated message size
 ...                                                   ]
 ...                                                   additionalHeader            /*  ExtensionObject                           */: null
 ...                                                 }
 ...                                                };`

mikakaraila, 2024-01-21:
Could be that client certificate must be trusted after update.

matteovi, 2024-01-23:
I do not think so. It worked until the day before.
It was resolved with a version downgrade anyway

mikakaraila, 2024-02-01:
Open each client, select endpoint, save and deployed flow, re-test.

closed 2024-02-01 by mikakaraila