mikakaraila/node-red-contrib-opcua #650: [NODE-OPCUA-E11] message size 65537 exceeds the negotiated message size 8192 nb chunks 1

fanbel, 2024-01-17:
we still face the same issue as https://github.com/node-opcua/node-opcua/issues/1263

we set up the settings for the connection as followed:
[image]

How did we even get there? Well we had the same issue in the logs as in the other issue described. So we tested the connection with the Softing UA Client and it worked, so it kinda had to be an issue with the client. So we fiddled with the settings in the Softing UA Client and found out, that the connection won't be established when the MessageSize is set to 8192 for example.

That's why we were quite sure, that if we set up the same MaxMessageSize in the Node-Red-OPCUA-Client that it would be working. WRONG -> see the following logs. There we see that the MessageSize is still capped to 8192. Is the setting really working? Is it a default error message, which doesn't acutally includes the current set MaxMessageSize? Is there a deeper way to debug the transport settings?
[image]

So in conclusion: it looks like the transport settings, especially MaxMessageSize, aren't registered by the opcua-client-node.

Node-Red version: v3.1.3
OPCUA-client-version: 0.2.320

Could you please help us out? Thanks in advance!

mikakaraila, 2024-01-22:
Hmm, there is new node-opcua 2.200.0 that contains something related to this...

fanbel, 2024-01-22:
Hi Mika, thanks for the reply and information. We looked into and tested it with a dedicated node-js-client entirely based on the linked node-js-opcua from Etienne and his team. But we discovered something interesting.
At first we tested with the new version 2.200.0 (v2.120.0) and it worked with setting the messageSize to something like 66000 (greater than the needed size).
But we tested the same with the old version, which is linked in your node from node-red-opcua (2.113.0) and it also worked. Also here we tested with a node-js-opcua-client. Here we also set the messageSize to 66000 and the connection worked. If we set it to 8192 like in opcua-node-red, then obviously it points to an error (see following screenshot).
[image]

So in conclusion it looks like that it's not a fault from the base library (node-js-opcua) but from the node-red-opcua implementation. It seems like the values we set in the UI of node-red doesn't actually get used in the connection. Therefore the error, where it states that the defined messageSize of 8192 is too small, even though we set it to 66000 in the UI.

Could you please check, if you kinda use the transportsettings (messageSize etc.) correctly for the opcua-connection? Thank you!

fanbel, 2024-01-22:
So i've looked into your code and found out that if we use transport settings, we should see it in the debug ->
[image]

But as you see, the client gets created with different transport settings (lower row). So from my point of view, the client gets created with the wrong transport settings. Can you verify this please?

mikakaraila, 2024-01-29:
I will publish first one major fix and take again look into this later...

TheeeEnding, 2024-02-14:
Hey @mikakaraila, I would also be interested in finding a solution for this, as I need to specify the MaxMessageSize in the client to establish a connection with my opc ua sever. Will you be looking into this in the near future?

Thanks in advance!

mikakaraila, 2024-02-14:
I will take a quick look tomorrow, now I have to rest.

mikakaraila, 2024-02-15:
Stupid errors, console debug looks now as it should be:
[image]

mikakaraila, 2024-02-15:
Test with the new version, re-open if not working.

closed 2024-02-15 by mikakaraila

fanbel, 2024-03-06:
looks good for now, thank you!