mikakaraila/node-red-contrib-opcua #651: 0.2.320 unusable in Node-red docker

dleidert, 2024-01-18:
We installed `node-red-contrib-opcua` version 0.2.320 in Node-Red version 3.1.3 running as docker-container (their official containers). The module shows errors and is not usable:

```
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:21)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:19)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:21)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:21)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:21)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:28)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:29)
Error [ERR_REQUIRE_ESM]: require() of ES Module /data/node_modules/chalk/source/index.js from /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js not supported. Instead change the require of index.js in /data/node_modules/node-opcua-alarm-condition/dist/client_alarm_tools_dump_event.js to a dynamic import() which is available in all CommonJS modules. (line:28)
```

mikakaraila, 2024-01-19:
Sorry, cannot help just now. Seems to be incompatibility with the latest nodejs and node-opcua with chalk.

mikakaraila, 2024-01-22:
And as I am not a docker expert I cannot help you, sorry.

dleidert, 2024-01-22:
Indeed, one of the modules must have installed chalk 5.0.0 (into the `/data` path).  I can see it. I manually downgraded it to 4.1.2 and everything seems fine. However, I cannot reproduce the issue. I started to install all modules we use in the affected container in a fresh node-red container and ended up with chalk 4.1.2 and not 5.0.0.

closed 2024-01-22 by mikakaraila