mikakaraila/node-red-contrib-opcua #652: Invalid Endpoint when connecting to Siemens S7-1500

AndreSchm, 2024-01-22:
I am using the following setup to connect remotly to a S7-1500
node-red-contrib-opcua 0.2.320
22 Jan 07:56:58 - [info] Node-RED version: v3.1.3
22 Jan 07:56:58 - [info] Node.js  version: v18.19.0
22 Jan 07:56:58 - [info] Windows_NT 10.0.19045 x64 LE
22 Jan 07:57:01 - [info] Loading palette nodes
using nodejs crypto (native)
Had previously Node14 running, upgraded to 20 → same problem, wanted to reinstall node-red-contrib-opcua → problem to install running node20 → installed node 18 instead

The PLC is as follows configured:
[image]
[image]
The client is as follows configured:
[image]

Using UaExpert I can connect to the server without problems.
In Node-RED I tried all combinations of Security Policies and Message Security Mode an checked the password multiple times without success.
I get the following output:
22 Jan 08:10:36 - [warn] [OpcUa-Client:78a1277482b6b074] : Creating OPCUA CLIENT
22 Jan 08:10:36 - [warn] [OpcUa-Client:78a1277482b6b074] : connect_opcua_client
07:10:37.223Z :client_secure_channel_layer   :236   requestData not found for requestId =  1 try with  2
22 Jan 08:10:37 - [warn] [OpcUa-Client:78a1277482b6b074] : Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [3]
22 Jan 08:10:37 - [warn] [OpcUa-Client:78a1277482b6b074] :         using securityPolicy: ["Basic256Sha256"]
22 Jan 08:10:37 - [warn] [OpcUa-Client:78a1277482b6b074] : Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: MyUser and password: MyPassword
22 Jan 08:10:37 - [warn] [OpcUa-Client:78a1277482b6b074] : Case C: With Sign you cannot use SecurityPolicy None!!
22 Jan 08:10:37 - [warn] [OpcUa-Client:78a1277482b6b074] : Invalid endpoint parameters:
22 Jan 08:10:37 - [error] [OpcUa-Client:78a1277482b6b074] Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"e5d7a8033ac069f5\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{\"user\":\"MyUser\",\"password\":\"MyPassword\"},\"endpoint\":\"opc.tcp://MyIp:MyPort\",\"securityPolicy\":\"Basic256Sha256\",\"securityMode\":\"SignAndEncrypt\",\"login\":true,\"none\":false,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":\"MyUser\",\"password\":\"MyPassword\"}"]
22 Jan 08:10:37 - [error] [OpcUa-Client:78a1277482b6b074] Invalid endpoint

mikakaraila, 2024-01-31:
Did you open each client, select endpoint, save and deployed flow?

AndreSchm, 2024-02-01:
Yes, I configured each client and deployed the flow. I even tried to delete all other opc flows and connections but without any difference regarding the Siemens connection.

mikakaraila, 2024-02-01:
Have you accepted client certificate on server side?

AndreSchm, 2024-02-01:
Have I to accept it in TIA? As shown in the 2nd screenshot "Automatically accept client certificates during runtime" is enabled.

AndreSchm, 2024-02-01:
Moreover the UaExpert Connection did't need any certificate acceptance on server side either.

mikakaraila, 2024-02-01:
Hmm, you are not using the latest version node-red-contrib-opcua 0.2.321

I fixed certificate managers to be singletons for client/user/server certificates.

Please update and test again.

AndreSchm, 2024-02-01:
I reinstalled 0.2.321, restarted NR, deployed a new flow but the error remains:
[image]

mikakaraila, 2024-02-01:
But now error message says exactly that Basic256 is not found from server.

AndreSchm, 2024-02-03:
Currently I have no setup to test. I will test during next two weeks and report.

closed 2024-02-15 by mikakaraila