mikakaraila/node-red-contrib-opcua #653: I have a problem kepSeverEX: invalid endpoint check that server has security policy: [none] on node-red

Platongsz, 2024-01-23:
here is debug :
23 Jan 17:36:43 - [warn] [OpcUa-Client:648938b243d31f74] :         using securityPolicy: ["None"]
23 Jan 17:36:43 - [warn] [OpcUa-Client:648938b243d31f74] : Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined
23 Jan 17:36:43 - [warn] [OpcUa-Client:648938b243d31f74] : Case C: With Sign you cannot use SecurityPolicy None!!
23 Jan 17:36:43 - [warn] [OpcUa-Client:648938b243d31f74] : Invalid endpoint parameters:
23 Jan 17:36:43 - [error] [OpcUa-Client:648938b243d31f74] Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"b92311036cf610bf\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://127.0.0.1:49320\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"]
23 Jan 17:36:43 - [error] [OpcUa-Client:648938b243d31f74] Invalid endpoint
23 Jan 17:37:02 - [info] Stopping flows
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : closing opcua client true userIdentity {"type":0}
23 Jan 17:37:02 - [info] Stopped flows
23 Jan 17:37:02 - [info] Updated flows
23 Jan 17:37:02 - [info] Starting flows
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : Creating OPCUA CLIENT
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : connect_opcua_client
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : userIdentity is ANONYMOUS
23 Jan 17:37:02 - [info] Started flows
10:37:02.529Z :client_base_impl              :550   Creating default certificate ... please wait
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : Case A: Endpoint does not contain, 1==None 2==Sign 3==Sign&Encrypt, using securityMode: [1]
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] :         using securityPolicy: ["None"]
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : Case C: With Sign you cannot use SecurityPolicy None!!
23 Jan 17:37:02 - [warn] [OpcUa-Client:648938b243d31f74] : Invalid endpoint parameters:
23 Jan 17:37:02 - [error] [OpcUa-Client:648938b243d31f74] Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"b92311036cf610bf\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://127.0.0.1:49320\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"]
23 Jan 17:37:02 - [error] [OpcUa-Client:648938b243d31f74] Invalid endpoint

Platongsz, 2024-01-23:
[image]

mikakaraila, 2024-02-01:
Open each client, select endpoint, save and deployed flow, re-test.

closed 2024-02-01 by mikakaraila