mikakaraila/node-red-contrib-opcua #656: Issue with the node red invalid endpoint.

zotqkf, 2024-01-31:
Hello Mika,

I want to try to connect the Ua Expert and Node Red, but it keeps failed.

1 Feb 08:19:57 - [info] Node-RED version: v3.1.3
1 Feb 08:19:57 - [info] Node.js  version: v20.11.0
1 Feb 08:19:57 - [info] Windows_NT 10.0.19045 x64 LE
1 Feb 08:19:59 - [info] Loading palette nodes
1 Feb 08:20:00 - [info] Dashboard version 3.6.2 started at /ui

It said   [31mClient node error on:  error: ["Wrong endpoint parameters: {\"id\":\"fb6b14efa86964fb\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{\"user\":\"admin\",\"password\":\"2d4fd419\"},\"endpoint\":\"opc.tcp://192.168.1.20:4840\",\"securityPolicy\":\"Basic256\",\"securityMode\":\"SignAndEncrypt\",\"login\":true,\"none\":false,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":\"admin\",\"password\":\"2d4fd419\"}"][39m

[image]
[image]

mikakaraila, 2024-02-01:
Did you accept client certificate on server side?

mikakaraila, 2024-02-01:
Are you using latest version?

zotqkf, 2024-02-01:
> Did you accept client certificate on server side?

Yes it is the latest version. I think I accept the one certificate on Ua Expert. Trust server certificate.
Where can I find the client certificate on server side?

mikakaraila, 2024-02-02:
Depends Windows/Linux, here Windows:
[image]

Standard PKI structure. Client trusted certificates:

[image]

zotqkf, 2024-02-02:
I checked the PKI folder, it was empty. Should I create the new certification?
I can I craete the new certification?

[image]

zotqkf, 2024-02-05:
Hello Mika,

I read the article and I knew what I did. I installed the openssl, and created client certificate using 
'openssl genrsa -out ./client_nd_key.key 2048
openssl rsa -in ./client_nd_key.key -text >client_nd_key.pem   
openssl req -x509 -sha256 -days 3650 -key ./client_nd_key.key -outform DER -out client_nd.der
openssl x509 -inform der -in client_nd.der -out client_nd.pem '

and put it in the server side. But still it say same error code.

mikakaraila, 2024-02-06:
I will check this at some day, just now no time.

There is also book that explains much more than I can:
https://leanpub.com/node-opcuabyexample-edition2024?gclid=Cj0KCQiAzoeuBhDqARIsAMdH14HqSRHXOkKbV9y6g3DGKjYdOf3p_NiISU2ovQkUw9YmIvwsFwJD1y8aAqkqEALw_wcB

closed 2024-02-15 by mikakaraila