mikakaraila/node-red-contrib-opcua #657: Quick Fix on Flows.Nodered.org broken link

gdziuba, 2024-02-01:
Hi Team,

I am doing some testing and saw that there is a broken link on the [flows.nodered.org page](https://flows.nodered.org/node/node-red-contrib-opcua).

[image]

The example hyper link got here: https://raw.githubusercontent.com/mikakaraila/node-red-contrib-opcua/master/examples
and should go here:  https://github.com/mikakaraila/node-red-contrib-opcua/tree/master/examples

mikakaraila, 2024-02-14:
Hmm, readme.md with the new link does not seem to work...

mikakaraila, 2024-02-15:
Of course... I will re-check this.

closed 2024-02-15 by mikakaraila