mikakaraila/node-red-contrib-opcua #658: client node  error

vnvasireddy, 2024-02-05:
hi,

when querying opcua server frequently getting error "Client node error on:  error: [" error at active reading: Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)"]" . Can you please tell what was the exact issue and how to solve it.

Thanks in advance,
vani

mikakaraila, 2024-02-09:
Your read interval is perhaps longer than session timeout or subscription...
More information needed to advice.

vnvasireddy, 2024-02-12:
here every 1 minute we are querying for the data...number of max data points will be 250.

[
    {
        "id": "1546237bd36b3e24",
        "type": "delay",
        "z": "739b7f8fb62fe88c",
        "name": "",
        "pauseType": "rate",
        "timeout": "5",
        "timeoutUnits": "seconds",
        "rate": "1",
        "nbRateUnits": "0.01",
        "rateUnits": "second",
        "randomFirst": "1",
        "randomLast": "5",
        "randomUnits": "seconds",
        "drop": false,
        "allowrate": false,
        "outputs": 1,
        "x": 650,
        "y": 1020,
        "wires": [
            [
                "53dc72a038dabf40"
            ]
        ]
    },
    {
        "id": "7aede8de9e74f3b2",
        "type": "inject",
        "z": "739b7f8fb62fe88c",
        "name": "",
        "props": [],
        "repeat": "60",
        "crontab": "",
        "once": true,
        "onceDelay": "5",
        "topic": "",
        "x": 170,
        "y": 340,
        "wires": [
            [
                "a64ee360c5ed677e"
            ]
        ]
    },
    {
        "id": "a64ee360c5ed677e",
        "type": "template",
        "z": "739b7f8fb62fe88c",
        "name": "config.csv",
        "field": "payload",
        "fieldType": "msg",
        "format": "text",
        "syntax": "mustache",
        "template": "s.no,cus_tag,iplon_tag,bd,dd,dt,fd,h,iid,m,p,qu,text,u,mf\n1,ILC-1-G1P4.ILC1-MFM-INC.Group1.R-Y phase Voltage,B10.INC_B10_MFM01.UAC12,B10,INC_B10_MFM01,INC_B10_MFM01,UAC12,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n2,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y_B phase Voltage,B10.INC_B10_MFM01.UAC23,B10,INC_B10_MFM01,INC_B10_MFM01,UAC23,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n3,ILC-1-G1P4.ILC1-MFM-INC.Group1.B_R phase voltage,B10.INC_B10_MFM01.UAC31,B10,INC_B10_MFM01,INC_B10_MFM01,UAC31,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n4,ILC-1-G1P4.ILC1-MFM-INC.Group1.R-phase to Neutral Voltage,B10.INC_B10_MFM01.UAC1,B10,INC_B10_MFM01,INC_B10_MFM01,UAC1,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n5,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y_phase to Neutral Voltage,B10.INC_B10_MFM01.UAC2,B10,INC_B10_MFM01,INC_B10_MFM01,UAC2,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n6,ILC-1-G1P4.ILC1-MFM-INC.Group1.B_phase to Neutral Voltage,B10.INC_B10_MFM01.UAC3,B10,INC_B10_MFM01,INC_B10_MFM01,UAC3,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n7,ILC-1-G1P4.ILC1-MFM-INC.Group1.Phase to Neutral Voltage-avg,B10.INC_B10_MFM01.UAC,B10,INC_B10_MFM01,INC_B10_MFM01,UAC,Server_8684,8684,1,Polepally_25MW,0,opcua,V,1\n8,ILC-1-G1P4.ILC1-MFM-INC.Group1.R Phase Current,B10.INC_B10_MFM01.IAC1,B10,INC_B10_MFM01,INC_B10_MFM01,IAC1,Server_8684,8684,1,Polepally_25MW,0,opcua,A,1\n9,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y Phase Current,B10.INC_B10_MFM01.IAC2,B10,INC_B10_MFM01,INC_B10_MFM01,IAC2,Server_8684,8684,1,Polepally_25MW,0,opcua,A,1\n10,ILC-1-G1P4.ILC1-MFM-INC.Group1.B Phase Current,B10.INC_B10_MFM01.IAC3,B10,INC_B10_MFM01,INC_B10_MFM01,IAC3,Server_8684,8684,1,Polepally_25MW,0,opcua,A,1\n11,ILC-1-G1P4.ILC1-MFM-INC.Group1.Line to Neutral Current,B10.INC_B10_MFM01.IAC,B10,INC_B10_MFM01,INC_B10_MFM01,IAC,Server_8684,8684,1,Polepally_25MW,0,opcua,A,1\n12,ILC-1-G1P4.ILC1-MFM-INC.Group1.R-phase Active Power,B10.INC_B10_MFM01.PAC1,B10,INC_B10_MFM01,INC_B10_MFM01,PAC1,Server_8684,8684,1,Polepally_25MW,0,opcua,kW,1\n13,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y-phase Active Power,B10.INC_B10_MFM01.PAC2,B10,INC_B10_MFM01,INC_B10_MFM01,PAC2,Server_8684,8684,1,Polepally_25MW,0,opcua,kW,1\n14,ILC-1-G1P4.ILC1-MFM-INC.Group1.B_phase Active power,B10.INC_B10_MFM01.PAC3,B10,INC_B10_MFM01,INC_B10_MFM01,PAC3,Server_8684,8684,1,Polepally_25MW,0,opcua,kW,1\n15,ILC-1-G1P4.ILC1-MFM-INC.Group1.Active Power-avg,B10.INC_B10_MFM01.PAC_AVG,B10,INC_B10_MFM01,INC_B10_MFM01,PAC_AVG,Server_8684,8684,1,Polepally_25MW,0,opcua,kW,1\n16,ILC-1-G1P4.ILC1-MFM-INC.Group1.R-phase Reactive Power,B10.INC_B10_MFM01.QAC1,B10,INC_B10_MFM01,INC_B10_MFM01,QAC1,Server_8684,8684,1,Polepally_25MW,0,opcua,kVAR,1\n17,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y_phase Reactive Power,B10.INC_B10_MFM01.QAC2,B10,INC_B10_MFM01,INC_B10_MFM01,QAC2,Server_8684,8684,1,Polepally_25MW,0,opcua,kVAR,1\n18,ILC-1-G1P4.ILC1-MFM-INC.Group1.B_phase Reactive power,B10.INC_B10_MFM01.QAC3,B10,INC_B10_MFM01,INC_B10_MFM01,QAC3,Server_8684,8684,1,Polepally_25MW,0,opcua,kVAR,1\n19,ILC-1-G1P4.ILC1-MFM-INC.Group1.Reactive Power avg,B10.INC_B10_MFM01.QAC_AVG,B10,INC_B10_MFM01,INC_B10_MFM01,QAC_AVG,Server_8684,8684,1,Polepally_25MW,0,opcua,kVAR,1\n20,ILC-1-G1P4.ILC1-MFM-INC.Group1.R-Phase Apparent Power,B10.INC_B10_MFM01.SAC1,B10,INC_B10_MFM01,INC_B10_MFM01,SAC1,Server_8684,8684,1,Polepally_25MW,0,opcua,kVA,1\n21,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y-phase Apparent Power,B10.INC_B10_MFM01.SAC2,B10,INC_B10_MFM01,INC_B10_MFM01,SAC2,Server_8684,8684,1,Polepally_25MW,0,opcua,kVA,1\n22,ILC-1-G1P4.ILC1-MFM-INC.Group1.B_phase Apparent Power,B10.INC_B10_MFM01.SAC3,B10,INC_B10_MFM01,INC_B10_MFM01,SAC3,Server_8684,8684,1,Polepally_25MW,0,opcua,kVA,1\n23,ILC-1-G1P4.ILC1-MFM-INC.Group1.Apparent Power - avg,B10.INC_B10_MFM01.SAC_AVG,B10,INC_B10_MFM01,INC_B10_MFM01,SAC_AVG,Server_8684,8684,1,Polepally_25MW,0,opcua,kVA,1\n24,ILC-1-G1P4.ILC1-MFM-INC.Group1.R-phase PF,B10.INC_B10_MFM01.PF1,B10,INC_B10_MFM01,INC_B10_MFM01,PF1,Server_8684,8684,1,Polepally_25MW,0,opcua,dec,1\n25,ILC-1-G1P4.ILC1-MFM-INC.Group1.Y_phase PF,B10.INC_B10_MFM01.PF2,B10,INC_B10_MFM01,INC_B10_MFM01,PF2,Server_8684,8684,1,Polepally_25MW,0,opcua,Jan,1\n26,ILC-1-G1P4.ILC1-MFM-INC.Gro [... cut ...]
                "3253c1b9abe9f210",
                "aeb40b0ba750dd19"
            ],
            []
        ]
    },
    {
        "id": "d27d372517d21e1f",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.10.2:49320",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]

above is the flow we are using to query fyi

thanks in advance

mikakaraila, 2024-02-15:
Could it be that you are not injecting read sometimes so that timeout is exceeded?
I mean if timeout is 15s and next read takes over 15s.

vnvasireddy, 2024-02-15:
"timeout" mentioned here  in server or client...if client, can i increase the timeout value, any possibility. Here every 1 minute data we are reading

mikakaraila, 2024-02-15:
Hmm, I expect there is default transactionTimeout used:
[image]

Proposal: Try to read just current time (ns=0;i=2258) every second or other interval less than 15 seconds.

vnvasireddy, 2024-02-20:
do all client nodes with **same endpoint**  will have timeout of 15 seconds

here we have 4 client nodes with same end point url  in 4 node-red flows which executes with 15sec delay one after another.
first client node will execute 0th sec and 2nd at 15th sec and 3rd at 30 sec and so on..so first client node will execute every 0th sec of a minute. 

any suggestion in timing where we are querying 1000 datapoints for every 1 minute.

mikakaraila, 2024-02-20:
Same endpoint and each have 15 second timeout, your current implementation will exceed it.

What are server limits? 

How many variables your can read in one request?

What is response time to get maximum variables in one request?