mikakaraila/node-red-contrib-opcua #659: need suggestion for creating opcua server

vnvasireddy, 2024-02-07:
please provide any sample flow to create opcua server with following details

1. create a folder having folder name "myFolder"
2. inside folder 2 objects 1. INV1 2. INV2
3. indside onbjects (INV1) variables 1. PAC, 2. QAC 3.FAC

when i tried i was unable to create objects. So please suggest for the above

Thanks in advance

mikakaraila, 2024-02-07:
Look examples folder and OPCUA-TEST-NODES.json
Tab node-red ua-server contains complete example for this purpose:
[image]

mikakaraila, 2024-02-07:
Of course "object" and variables of custom type can be problematic...

mikakaraila, 2024-02-07:
Try to use example as it can create folder and variables inside it. It is the way flow based programming is meant to be used.
JavaScript inside function node is not flow programming...

vnvasireddy, 2024-02-07:
ok sir..will follow the suggestion. thank you

closed 2024-02-08 by mikakaraila