mikakaraila/node-red-contrib-opcua #660: Node Red OPC UA Client does not close connection each time I activate the deploy

OfirLevi888, 2024-02-07:
Hello,
I'm using Node-Red OPC UA client to connect OPC UA server device.
The maximum number of OPC UA connections this device support is 5.
It seems like each time I make a change in the flow and deploy again the project, Node-Red OPC UA Client is not closing the connection but opening a new session.
This results that after several deployments, I have to reset the OPC UA server device in order to enable again the OPC UA sockets.
1. Do restart flows, or deploy actions should close the existing OPC UA connection?
2. Is there a way to initiate OPC UA client close connection manually?

Thanks for your help!!!

mikakaraila, 2024-02-08:
Use action disconnect

OfirLevi888, 2024-02-08:
Which block should I use for disconnect?

mikakaraila, 2024-02-08:
Function node is enough just look example

mikakaraila, 2024-02-09:
Here example (tested):
[image]

OfirLevi888, 2024-02-09:
Many Thanks MikaKaraila!!
I will test it.

OfirLevi888, 2024-02-19:
Hi again,

I'm implementing a method call in OPC UA.
Which type of action should I use for this method? Can I use the subscribe or should I use the read?
Can you explain the differences?

mikakaraila, 2024-02-22:
Simply use method node, there are examples inside OPCUA-TEST-NODES.json