mikakaraila/node-red-contrib-opcua #662: Security token haven't renewed

RudolfTheRedNode, 2024-02-08:
Hi @mikakaraila and All,

I got an issue my connection to opc UA suddenly got error like this. can anyone help me regarding this issue.

[image]

Best regards,
Kenneth

mikakaraila, 2024-02-09:
Are reading with interval that is more than token / session timeout?
A bit more information when this happens would help to give more advices.

RudolfTheRedNode, 2024-02-15:
The interval for request is every 1 second, using the trigger node.

mikakaraila, 2024-02-15:
Hmm, still not enough context to help you...
Is this happening immediately or after some minutes?

RudolfTheRedNode, 2024-02-19:
Hi @mikakaraila,

Sorry to let you think hard, it turns out that the problem is the device not responding well ^^,

But I got a new scenario now that after reading so many nodes, I got a delay of around 2s to 4s then it will read again.
Is this normal or a bug?

mikakaraila, 2024-02-19:
Hmm, it can be that device cannot response faster. One reason I normally use subscription to get just updated values.

RudolfTheRedNode, 2024-02-19:
Hi @mikakaraila ,

Thanks for advise but now I get this message from the console.

09:01:17.723Z :client_session_impl           :709   [NODE-OPCUA-W21] Pending transactions:  ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest ReadRequest
09:01:17.724Z :client_session_impl           :710   [NODE-OPCUA-W22] Warning : your opcua client is sending multiple requests simultaneously to the server ReadRequest
 ...                                                  please fix your application code
 
 What could be the solution for this?

mikakaraila, 2024-02-19:
Are you using multiple client nodes? Use just one! 

Now you inject multiple parallel read requests to server.

RudolfTheRedNode, 2024-02-20:
Hi @mikakaraila,

This is my flow shown below. then I got the error which is shown below:

[image]

20 Feb 16:51:01 - [error] [OpcUa-Client:AMS OPC Client WRITE] Client node error on: AMS OPC Client WRITE error: ["Write multiple as array session is not active!"]
20 Feb 16:51:01 - [error] [OpcUa-Client:AMS OPC Client WRITE] Client node error on: AMS OPC Client WRITE error: ["Write multiple as array session is not active!"]
20 Feb 16:51:01 - [error] [OpcUa-Client:AMS OPC Client WRITE] Client node error on: AMS OPC Client WRITE error: ["Write multiple as array session is not active!"]
21 Feb 04:11:15 - [error] [OpcUa-Client:AMS OPC Client] Client node error on: AMS OPC Client error: ["AMS OPC Client error at active reading: Connection Break"]
21 Feb 04:11:15 - [error] [OpcUa-Client:AMS OPC Client] Client node error on: AMS OPC Client error: ["AMS OPC Client error at active reading: Invalid Channel BadConnectionClosed"]
21 Feb 04:11:15 - [error] [OpcUa-Client:AMS OPC Client] Client node error on: AMS OPC Client error: ["AMS OPC Client error at active reading: Invalid Channel BadConnectionClosed"]
21 Feb 04:11:15 - [error] [OpcUa-Client:AMS OPC Client] Client node error on: AMS OPC Client error: ["AMS OPC Client error at active reading: Invalid Channel BadConnectionClosed"]
21 Feb 04:11:15 - [error] [OpcUa-Client:AMS OPC Client] Client node error on: AMS OPC Client error: ["AMS OPC Client error at active reading: Invalid Channel BadConnectionClosed"]