mikakaraila/node-red-contrib-opcua #663: opcua client not connecting

vnvasireddy, 2024-02-08:
hi,
we are using servers having OS hardening.
in one server opcua client getting connected to server without any issues, but in another server it is throwing error.
any suggestion regarding errors
here we are not using any security..trying to connect as anonymous.
[image]
[image]

screenshots attached for both servers

mikakaraila, 2024-02-09:
Look console and turn on logLevel debug.
Could have something to do with permissions like access to certificate folders (PKI).

vnvasireddy, 2024-02-12:
removed docker container and ceated a new one and installed opcua again, solved issue. Thank you for quick response

closed 2024-02-12 by vnvasireddy