mikakaraila/node-red-contrib-opcua #667: MONITORING BOOLEAN DATA ITEMS

carlosbama3, 2024-02-20:
Hi Mika,

I want to monitor node of boolean types, but i can't. However with susbcribe mode i can it.

It could be monitoring boolean nodes or it is not possible?

Regards.

mikakaraila, 2024-02-21:
It is just as any other variable, look OPCUA-TEST-NODES.json from the examples folder.

carlosbama3, 2024-02-21:
I saw them, but I have read that in some opc ua servers with booleans variables when the opc ua client request the values of these, in this case True (1) or False (0), the deadband (in our case is 0 absolute) dificults to get the boolean values because the conflict with this values.

Are this true?

I'm going to read again the opcua-test-nodes, but in my case the opc ua server has variables with all types.

Regargs

mikakaraila, 2024-02-21:
Should not be a problem

carlosbama3, 2024-04-24:
HI Mika, we are checking your test nodes examples. But we don't see anything, do you sure that we can monitor the boolean types? With subscription mode only we have the first value, later we have keepalive and with monitor mode We don't even have the first value.

mikakaraila, 2024-04-24:
Works as expected, you have something wrong in your flow.
[image]

carlosbama3, 2024-04-24:
Look, my values of boolean are not True or False, they are 0 or 1:
<img src="data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAooAAAC0CAYAAAATmw14AAAAAXNSR0IArs4c6QAAAARnQU1BAACxjwv8YQUAAAAJcEhZcwAADsMAAA7DAcdvqGQAAP+lSURBVHhe7L0HYJRF/v//2pq2qYRAAoHQe0eqYkERrIdi96xnQc/e29lQRBQ7oFhAQFSa9Cq9907oBEiBlM2WbC//mWc3JCio+L3f/c9jXjDZfT7PzDxln2fmPV0XFqBQKBQKhUKhUPwCffRToVAoFAqFQqE4BSUUFQqFQqFQKBSnRQlFhUKhUCgUCsVpUUJRoVAoFAqFQnFadH6/Xw1mUSgUCoVCoVD8Ct2JEyeUUFQoFAqFQqFQ/ArdypUrlVBUKBQKhUKhUPwKXUVFhRKKCoVCoVAoFIpfoSsvL1dCUaFQKBQKhULxK9SoZ4VCoVAoFArFaVFCUaFQKBQKhUJxWpRQVCgUCoVCoVCcFiUUFQqFQqFQKBSnRQlFhUKhUCgUCsVpUUJRoVAoFAqFQnFalFBUKBQKhUKhUJwWJRQVCoVCoVAoFKdFCUWFQqFQKBQKxWlRQlGhUCgUCoVCcVqUUFQoFAqFQqFQnBYlFBUKhUKhUCgUp0UJRYVCoVAoFArFadHt2rUrHP1O7dq10ev1hEIhjh8/Tjh8cpdCoVAoFAqF4hxD9+abb2pqsFGjRlxzzTUEg0GcTieTJk3C4XBonhSK/3U8Hg+xsbHRLcVfEbfbTVxcXHRLoVAoIlRUVJCQkBDdOrf5MxWAuvz8fC2UrElMS0vTMkwpFMeOHUvv3r2pX7++5lGh+F8lEAgwd+5cevToob0Dir8e5eXlLFmyhAEDBmi/p0KhUEh8Ph8zZ87kyiuvJCYmJmo9NzGZTFqB+mzFok4ksCdDyNJ4pVD89ttvufTSS2ncuLG2T6fTnYz8XGiSls3vinMDm83GxIkTufHGG4mPj8doNEb3KP4q7Nq1i61bt/L3v/9dyxhkwVehUChkN7rp06dzzz33aNrlXE7fZa2qTB//nwhFKRIrORdEoqT6NSv+t5FCccKECZpQtFgsmM3m6B7FX4Xt27ezZcsWTSjKQp4SigqFQlJUVMTUqVM1oSjTBVmrdq5SqfHOVsf9bmr6S8GkBJTiXMNlL6O0tFRzZaVWnI4KIUZ+/0Xzueyi9OaPbp0Z+dI6Hc6Tx6h+rD8SXnF6QgEv5TbxW0W3K5F2u80h7nvU8Atk04zX64tuKRSK/zVkYdJaZv1VmmstK9cqyvz+f3/3FdklptxqO2O689+M4fnnn38t+l1T2vJiZNWkbMaRA1wq+2xVF4jy+/+6U5w7eL1eduzYQatWrbTaRIPBEN2DNrhr1rt38sKoxeRu28DyVevZtmUT+eVBmtTPwmg6fTOGDLdk9BuM3mrg/PaNMerPnDroQz62LJzI1+N+ZP6sKbz70Sg279rP+nXrccfXpXm99HP2mZQD6uRv8nvXf+LECa3moF27dprwlt7LD6/lzVfext+wO80yKjuyhzm2aRJPv/kjbbp2p0biqbXH8nfbOHEoHyxx0KtjE8xGlRYoFH9lpPDLzc2lQ4cOWjoi03eX/SDff/QJUxYs4vtvRjJ62lItfd+w5Riu8jz22mJpWjcNvf7f9/678jby0gsjqdm9B3Us//+0WlVqvLPlrNpnKkXUv8v9XzhdfP9Op1BU4g8b6Nb/Id5//32GvvMGD990MceWf8v3S/dopUNZsLJaI6VTuz1SUxXyubQBFseKSqjwBjUBYrfZNT9loiQbCASjsQvpYoihy1V3Mvi9D3jvqVupX78dDw96j/ffHUSfDnVx+yMiU5aCbeU2PB6vSPwqNBEl43O7ZVOC5gW/z6+VlP8XaiMrp+mS9+7s0ZGa3ZxOdXxMn7dNyMMI4aCb9fMXENO4A1lp8VSI+1hZmyDvo+bH78bqcOP3+3FVuDWbRI6crKxpkL95WWmZdp/9PjV4RqH4K5GQ3Ij7XxvMsHff5q7eTWnX926GvP8Rb73xT/pd1Z+ruzUW77oPl8st0nS7VtMo02z5/Zfpd2Ua4rA7T6bDMs2qzBMqKlyRfMJlp8Im0hpbhbYtm4DLyiKtVTI91/INkeY5RB7iqxDHEWlLhTekVWRIP/JcJDKc/C67TEm7zA/+X3PWNYrnSh9FxbnDb9Uoyud917LJFKd156K2dcQ+PbE1sskKHODLuQWc36c5G8d/wNAxc1mzcgkLFyzAntCIeuHDfDt+Aut3HqZ2k/akFy/jzY/GsHTZMlYsXsDmwiDtWjQixlR1LEmwZD9Tl+6n5xV9qWtwMOeLV/hmU5hLOzfGU7iLV575F/srjvH5O5+zZs9WFs2dz+btu0lv1IE0k5fF33/MV19NZvna1azbXkwd8Q7/stbsr4JMaGUzsJzfVXZA/y3B+OsaRZFeGeJJpJC5s7bRum9v0sVtCJRt5pORC7jglodonpjHZ0L8T5m3nNVLF7BiYy71mrbFfXAVG2y1aBG7iyEfLqZjj47EGmHeqNf4ITeW85rUYPkYcZ+/ncKKdWvYsLOUuo0bk2Y5d/s+KRT/rZyuRvEkoQD71y/ggKE5l3ZqhEEUJBd88zbf7TJRs2gBLw76hsMHtjN12gy2FvjJ3zyXsT9MZV2hkS6tGuLYv4ov3/+QaT8vY9WyTeT5k2iRncL22V/w2Tc/sHzlauat3UtSrXocXTeNMVMXUW6KpU2Tmiwb9xEjJi1h5dKFLPp5EeY67agVu4dX73uJNQfymD1zGvPWH8R5YjMzps5g+spcmrVsx4nlX/LMWyPYtGsTixcsZueBfJFutSM57tS85HT8Xo2iTDvz8/NJSkqKWiKcVY3if0IkymP80ikU/23UzG6Cp+Qw+fYQiU178eKrbzBs2DAeuiyHKT/Mxle/M/3Ob0+Pa+6l/0U5hJLqc8MDz/LBBx/w2mN3cnj+bNYfKxHPdzTC0xAyJ9GhbQvyt27geFBH2YldlAaTaJqVhM/u57LrHtVKxJ2TjjNy4lL2r57BNzP3cPMzb/Dumy+Q6VjEh98vw+X7671D8r2XpeXMzExNIMoE/s8MUKnbuS/NEg6xZHORiDTEofWrsNbpyKXtstDp0+h55e289957DB38JjmOXXw7fwfeUKRQbAj7tRqEk2mQx4qtwsuBDZP4evFhbnvhLQa/9iyWE/P4ZOJK3Ko7qULxl0fvs2F3+fDpfbgDKVx59zO8elMbls1fRtPLH+TtR24gb/lcDhcd5sevPuJQZm+GvPsuT93WgaWjP2T79h1MmrqC+n2f5M033+Sh/t3ISo2j+8X96NK2C3c+cCf1k5Oo3aIf/3rjbS1PuKmlhW+mLKFMpDWuMheNL7iBD166H8/uVZQn9uD555+nsW0rCzcewKvz4g2mcv1dLzPk9RdIL1vHmPnbT6nl/DPIFpSFCxcyatQodu7cGbVG+MMp7y8FW6WI+3e703E6f9Xd2fB/Da9QVMekS9RqsfyF25g9ezYb9uRTcbwUl75qCIUuGEftxu1pEGtl0fxZzFm5RWu+sHl/u4uDFEe123enXnA/a3c52L9hPZY2vWiYGIeldmPa5KRjikukeZscKvZsYNe2TZQKcXlw20pmLlyhCZ4je7fi9Pz1mkZls4pETucg31HZJHNKTcAfxBhflws612HtrHmcKDnGnNmradK9H7UtEJ+aRduGNdmxZgE/zV1CnjimrH34rTRBnkfh7l3YzPHs37ycWT+vRKQif9n7rFAozkxidjY5NWIwxCQQa0khs2YqxlgLMW43ZfYCtuT7iDd4mT9/Hqv3l2Pw5bHxhJnWresy8+M3eG3o5xwq9pBqOXUhAENCDdp3bk7Z3pVavrHl2HHcR62RdC45gZy6tbX0Ti4CkdmkofaZFG8WBdXIIDtLdhva1LUQl5hKu8YZ5O3fI8Tjn+8yJ9O8BQsWiOuYr6W9csGVPXv2RPf+hlCsbGo+Hf9t4kqezy/db9l/yR/1p1BI5PNRcOQglvS6pJmKGDd4KN8u2KH1X5P9DX8paIK6Ctb+NJJ3v5pBsSMQbf74Y2U0U1JTurSwsPLnmSzcnE/Hrj2IM536bhoDxpPPbGViIY9Rp/2l3Hfr1aTFn31N3P+fyMSyuLhYq02sRJZ2/8z8ZzqDmXaiJG/OX8HGLRvZZk+mX8+mYk+Yoh2LGfTyp6zZb9NqKw3hP7Zygxx8VKkJZbisTldyz01X/OXus0Kh+D/id2vpgSQck0yfmx7kgg5NGPDQK7z+wl10qW/ihw+f4/2Jq3H5qioPPNbdDH9jGNPWHdVqAmWe8WfStwBGQnozJl0kX/kzVNYkSqEop4fr2bOn1r9+/PjxJ2sWz5iyyYxHHlgmhDrpxPdKJ22/5U7xrz+z/+px/tKdzv/pnHasau6k/ffi+J39CsXpkCLGfmAtE2fvoNcVfahZtJ/l+/O56Kqbufrqq2hVJ17zI5GrAMj+vkGbk3Xr1pHW+lJuvO4aLunQgNg/WvlkjBXisCfFS0eywl6fnu3StGfbU1REwQkrQW8FW9bnktK8A63adaJlahxNOvTiyst7oy/aytKdJ7TBOH8lXC4XNWrUICUlRetTU+lkmnT2c1zqqJHVkva1gowfNYn4xhfSKjNR+42KcleTl1SHW27oz5UXdiL9Fz+KrDkw6O2Ue3WESg9w8KBVu/e1W3egRUosrTr15Jo+vTDkr2NN7nF8IZVuKBTnCobkOlzUvj4JGY25/PK+XNIilWWr1mF1W9myPZecFh248dY7uPaitng9Xi0NE8kHIVcI96FdbCiq4JrrbuSaKy4jO9X4m30Hf4nryEFOlNnRVRxn87Yicpq0IMbw5yq4du/erdUkJiYmcvPNN2urW/Xr10/rHy5rFuUgvt9O2SrFlPisFHBS+FUXZL90mkCrDCedLmKLhIvGEXURP6ezVTq5HY3vNE7zL+Kv7s54bnJ/tW1NWFbGVc1e6RSKSmQN3ubpn/HUU0/x3LPP8+G4ebS++g7+fnFjDLVb07drBlM/epYXX3iRaSv2as2l8hlq0rENoZ0T+XLeLnpf2psTK77h6aef4oNvfsIdEytKgafpU2KM0zoSV+6Tz2jdNheQnRggq1VPmiVFhJLfcYKfRg/TzmlRcSb333QZTXpcwy2X1mfyh29q9sVFqdzet5M2EOOvhEywatWqpV17dWRfxT8ztQMJtejT5zyKKxx06n0JKTGRd75Bh0towW7efuUZXn37A3LzddpvbYpLIiHGQK3GF3HV+VkMf/8t3vhiJnmheE2oNj7vBm45vz7fCftjTz3HyrJa3Cji/6vdZ4VCAeb4pFNaaeS2fJdlc3NiXGSAms4UR4olVkuTZM2fLMTq4tO48vZ/krhlOs88/TQvfTKJpudfS+c6aVTkbuCTN97gkWdeYZ61Cddd3p0at [... cut ...] KF6TZCqSof4HotRCx6HjaApSpmtHPS4qLJXyxIQGWPBbJQrXYfwu6qprHajN5pFmgCaX8SHiKg4oswyvor/TTirynDWdjkI6CLMxEZFKX45JTyOKhxuH1p9BNGxUeJ+CAf8gg8FsIp7Niqu9j4Jh2t0BhEeg0FkGBTPUFWVU7n+JqMJn9hvTJ0ovNZKREIiRcKQeK4qqzxEJ1gU/10q/nsEfdXivLqU5zA6PgaTeJ+pUKFCRSjgxWa1CS2hJSImFkuNYSbgdeGwSY0heIuFqAijqIsDSndurKjLRUq84h1uk6ZDoUNi4mIwKpMzwrxD8EJZEKXU+zp8bgc2v5E6lrDByWmvxOnxo4+IIs5S27/8a3wQV5UVIuMUjSBdqNltTqFAdBjMFmXSjdxrwCN4+6/zVsFL1WO2xBBp0op4QrcpY9z+WcD/B0W4A8UMcm2iAAAAAElFTkSuQmCC" alt="imagen" iscopyblocked="false">

carlosbama3, 2024-04-24:
But, our boolean values are 1 or 0, is there any probem with this?
[image]

mikakaraila, 2024-04-24:
If Data Type is Boolean then value should true or false not 0 or 1.

carlosbama3, 2024-05-28:
Hi Mika, I continuous having the same problem...I'm a Simulator mode in Kepware server OPC UA. The documentation about boolean type is that:
[image]
So that, the value of boolean type is B0000-B9999
And if I configure that, the opc ua client can not get the values:

[image]

carlosbama3, 2024-05-28:
The only way to get the values in boolen type is like this:

-First: Establish the data type in short type (for exameple)
[image]

[image]

-Second: Change the data type to boolean
[image]

[image]