mikakaraila/node-red-contrib-opcua #668: Client can't connect to OPC-UA server and crashes Node-RED

markomedin, 2024-02-21:
Hi there,
I have been trying to solve this by myself but I figure out what is the cause of the error.
This is the error message, right before Node-RED crashes:

21 Feb 09:38:23 - [red] Uncaught Exception:
21 Feb 09:38:23 - [error] Error: 
    at assert (/Users/markomedin/.node-red/node_modules/node-opcua-assert/source/index.ts:12:21)
    at new ChunkManager (/Users/markomedin/.node-red/node_modules/node-opcua-chunkmanager/source/chunk_manager.ts:147:15)
    at new SecureMessageChunkManager (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/secure_message_chunk_manager.ts:107:29)
    at MessageChunker.chunkSecureMessage (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/message_chunker.ts:155:30)
    at ClientSecureChannelLayer._sendSecureOpcUARequest (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1813:29)
    at ClientSecureChannelLayer._internal_perform_transaction (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1618:14)
    at ClientSecureChannelLayer._performMessageTransaction (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1541:14)
    at ClientSecureChannelLayer._open_secure_channel_request (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1144:14)
    at Immediate.<anonymous> (/Users/markomedin/.node-red/node_modules/node-opcua-secure-channel/source/client/client_secure_channel_layer.ts:1241:18)
    at processImmediate (node:internal/timers:478:21)
    
    My PLC has as a setting Basic256Sha256, Sign&Encrypt and username&password. Generated certificates selfSigned.pem and private_key.pem and placed them in a Desktop folder. 
    In the Client node set for subscription, I defined the endpoint, the settings I mentioned, and specified the certificate paths as Local file certificates. So it has all the information. I can connect with Prosys Browser software, I can also connect with a python script using all the settings and certificates I mentioned, I can even run this python script as an exec node, but I can't get the OPC-UA Client node to run.
    Can you maybe give me some insight? Thank you.
    
    I am running my setup on a Mac. Using Node-RED version 3.1.5, and node-red-contrib-opcua version  0.2.323.

mikakaraila, 2024-02-22:
Looks for me vey low level problem. Are you using standard settings, I mean no transport settings modified?

What is server you try to connect?

mikakaraila, 2024-02-22:
Turn on logLevel debug to see a it more. Latest version uses certificate manager that creates needed certificates.
It can be that on Mac file system permissions are limiting PKI folder structure creation.

mikakaraila, 2024-02-22:
Edit settings.js under .node-red folder, this part:
[image]

Version before this merge allows to use own certificates:
[image]

markomedin, 2024-02-22:
I got it to run. Based on the PKI folder paths in the debug message, I found out that it creates the folders in node-opcua-default-nodejs. I assumed that they might be corrupted or something from past use, so I deleted the entire folder. Restarted Node-RED and it worked.

Can you tell me, now we cannot use user generated certificates but only this automatically generated ones?

22 Feb 10:19:36 - [debug] [OpcUa-Client:74025646cda251b2] Trusted folder:      /Users/markomedin/Library/Preferences/node-opcua-default-nodejs/PKI/trusted/certs
22 Feb 10:19:36 - [debug] [OpcUa-Client:74025646cda251b2] Rejected folder:     /Users/markomedin/Library/Preferences/node-opcua-default-nodejs/PKI/rejected
22 Feb 10:19:36 - [debug] [OpcUa-Client:74025646cda251b2] Crl folder:          /Users/markomedin/Library/Preferences/node-opcua-default-nodejs/PKI/trusted/crl
22 Feb 10:19:36 - [debug] [OpcUa-Client:74025646cda251b2] Issuers Cert folder: /Users/markomedin/Library/Preferences/node-opcua-default-nodejs/PKI/issuers/certs
22 Feb 10:19:36 - [debug] [OpcUa-Client:74025646cda251b2] Issuers Crl folder:  /Users/markomedin/Library/Preferences/node-opcua-default-nodejs/PKI/issuers/crl
: Create session with userIdentity node.client false userIdentity {"type":1,"userName":"test","password":"test"}
22 Feb 10:19:37 - [warn] [OpcUa-Client:74025646cda251b2] Create session with userIdentity node.client false userIdentity {"type":1,"userName":"test","password":"test"}
22 Feb 10:19:37 - [info] [debug:debug 30] 
{
  error: null,
  endpoint: 'opc.tcp://127.0.0.1:4840/SimulationServer',
  status: 'session active',
  _msgid: 'd4d0a8460e31bd3d'
}

mikakaraila, 2024-02-22:
It was actually some other problem that Etienne found and made PR that uses singleton for certificate manager.
I will try to look some workaround later, but not immediately... sorry.

closed 2024-02-22 by mikakaraila