mikakaraila/node-red-contrib-opcua #669: Handle Errors / no status

Test012345678949, 2024-02-22:
Hi,

maybe I can't see the solution but how can I handle if there are no items in the opcua client for multiple reading:

[image]

The Error which is thrown is not catchable I think and no bedug messages is sent
and the status isn't changed too.

If the error is catchable or the status will change to "Error Nr 123" I can handle an injections of items and the flow would be fail save or am I on the wrong way ??

Greetings

Test012345678949, 2024-02-22:
Thx for your immediate reply.

Youre right the latest version has 2 outputs, but the "No Items Error" doesn't start a message either:

[image]

mikakaraila, 2024-02-26:
Ack, I will fix this to next release and check other places at the same time.

Test012345678949, 2024-02-26:
Thx.

Yes normally I would say "That's right", but in the flow "programming" I have two or three times seen, that the OPCUA has no item's without been cleared. I think this is from partially restarting the flow.

To solve this issue permanent, my idea was: If no items are there --> add them and start the reading some seconds later.

Therefore the message "No items" has to be readable by the flow.

Or am I wrong ??

mikakaraila, 2024-02-26:
OK, this clarifies problem for me. I will be travelling this week. So most probably I will have time next week to fix & test this.

Test012345678949, 2024-02-26:
No problem.... You make great work and it runs stable and flawless.

I'm really happy for your short time reaction  and therefore it is no problem waiting 1 week.

mikakaraila, 2024-02-26:
I found some other similar cases and fixed other set status cases. 

I will do some testing before publishing new version...

Test012345678949, 2024-04-12:
Hi,
Do you know, when a new version will be published ???

mikakaraila, 2024-04-12:
Thanks to remind me. Have been busy and bit sick.

mikakaraila, 2024-04-12:
I just made & published new version. Not fully tested, so I will fix if there is something...

Test012345678949, 2024-04-12:
It runs good.

If status = no items to read --> add items --> start 5 s later new read.

I think there will be no bug within deploying and getting the itemlist cleared, cause then it automatically add the items after the error occurs 

Thx

mikakaraila, 2024-04-26:
I will close this now...

closed 2024-04-26 by mikakaraila