mikakaraila/node-red-contrib-opcua #671: [red] Uncaught Exception - (Cannot convert undefined or null to object) when Write with OPC Item of type ExtensionObject

BenoitFocant, 2024-02-29:
Hello,

I try to write à ExtensionObject from a MQTT topic to an OPC UA variable.

When I READ the variable using the same OPC Item there s no problem. 
But if I try to WRITE, there is an exception and node-red crash.

The Node-Red Log:

29 Feb 09:12:13 - [warn] [OpcUa-Client:344b29e6a347be4d] Create session with userIdentity node.client false userIdentity {"type":0}
29 Feb 09:12:24 - [red] Uncaught Exception:
29 Feb 09:12:24 - [error] TypeError: Cannot convert undefined or null to object
    at Function.assign (<anonymous>)
    at build_new_extensionObject_dataValue (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1768:40)
    at write_action_input (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1738:17)
    at OpcUaClientNode.processInputMsg [as _inputCallback] (/data/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:829:11)
    at /usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:214:26
    at Object.trigger (/usr/src/node-red/node_modules/@node-red/util/lib/hooks.js:166:13)
    at OpcUaClientNode.Node._emitInput (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:206:11)
    at OpcUaClientNode.Node.emit (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:190:25)
    at OpcUaClientNode.Node.receive (/usr/src/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:499:10)
    at deliverMessageToDestination (/usr/src/node-red/node_modules/@node-red/runtime/lib/flows/Flow.js:800:40)

NODE_RED_VERSION=v3.1.3 (docker image)

[image]

My Flow:
[image]

OPC item Config:
[image]

OPC Client Config for write operation:
[image]

The Value in the MQTT topic OPC_TEST/data:
{
  "mandatoryInt32": 0,
  "mandatoryStringArray": [
    "op",
    "hihaaaaaaaaaaaa"
  ]
}

What I Publish in MQTT to be write in OPC (view of debug 4 and 5 in my flow):
[image]

Thank you for your help :)

mikakaraila, 2024-03-04:
Hard to debug and test as I don´t have environment. This has always been problematic with custom types / extension objects...

I will try to look my code, but cannot promise anything.

BenoitFocant, 2024-03-04:
Hello, 

After investigation in other topics, I finaly found the good way to write ExtensionObject.
The topic [Can't instantiate Extension Object](https://github.com/mikakaraila/node-red-contrib-opcua/issues/605#top)
#605] was the most adapted for me.

Then, I think this topic can be closed :)

Br,

closed 2024-03-04 by BenoitFocant