mikakaraila/node-red-contrib-opcua #672: Bad too many connection error

frankTopspin, 2024-03-06:
Hi, I'm a newbie in using this plugin.
I've made some nodered flows to read/write some data from an OPC UA server (panasonic PLC).
Everything goes well, but something some crash occurs.
 When i try to relaunch nodered I give the message (from nodered debug) too many bad connections.
I've to relaunch nodered or restart machine more times to get it working again.
What can I do to fix this delay?
Sorry for my english.
Thanks

mikakaraila, 2024-03-07:
Use less client nodes, you cn use just one and inject action it will overwrite select action.

FrankRobot, 2024-04-16:
Hi mikakaraila, there are any news about that? There is a way to send multiple requests with the same connection without use multiple read?
What do you mean about "you cn use just one and inject action it will overwrite select action?"
What is the best practice to do that activity?
Can you give us some hints or some short example code?
Thank you very much!
Frank

mikakaraila, 2024-04-16:
Look examples and OPCUA-TEST-NODES.json and this tab:
[image]

FrankRobot, 2025-03-25:
Hi Mika, recently I'm working on another similar project and I have many difficults to use multiread and multiwrite, because the flow in nodered is very fragmented, and it is positioned in various tab. It is organized to get a value at a time and to launch different action, get other variables, or write something in other variables of the flow. In other words I don't know how to modify all the code to minimize the connections. There is a simpler method to avoid this error "bad too many connections"? There is a way to automatically or manually close the connections before reach the limit? There are another way to call the opcua without generating a new connection every time? Thanks in advance. Frank