mikakaraila/node-red-contrib-opcua #675: invalid channal BadConnectionClosed

PLCs7, 2024-03-20:
Good morning, I'm having problems with the opc-ua client node which has given me the "invalid channal BadConnectionClosed" error several times and i can't read variabiles from the server (with UAExpert it works) below the screen and when I do full deployment the connection is restored correctly.
This happened to me on several opc-ua servers, sometimes I also didn't receive any errors and had the green dot under the opc-ua server (active reading) but was unable to read the variables from the opc-ua server

Thank you in advance

version
node-red -->v3.0.2
opc-ua -->v0.2.323
[image]

ericsimob, 2024-03-21:
I have the same problem and also this message in the log:
[image]

mikakaraila, 2024-04-12:
I will update to latest node-opcua in next release as it contains fixes.

mikakaraila, 2024-04-12:
I just made & published new version. Not fully tested, so I will fix if there is something...

mikakaraila, 2024-04-26:
I will close this now... reopen if not yet OK

closed 2024-04-26 by mikakaraila