mikakaraila/node-red-contrib-opcua #677: Reconnect problematic of opcua client

TheeeEnding, 2024-03-25:
Hey Mika,
unfortunately I noticed a change from v0.2.320 to v0.2.321 that resulted in some opcua servers to not reconnect correctly after a downtime of the server (they show an active session on the UI but the node-red opcua client is getting no new messages from the server).
I'm talking about this keepSessionAlive line that you commented out because of the issue #599 and reactived in v0.2.321.

[image]

Would it be possible to comment the line out again or another workaround for me? It seems to mess with the keepAlive of the opcua client and I can't go back to the v0.2.320 as it does not include the correct configuration of transportSettings that another opcua server needs...

Thanks a lot in advance!

fanbel, 2024-04-05:
Can confirm that we have the same issue again... would be glad if you could take a look, thank you!

mikakaraila, 2024-04-12:
I will update to latest node-opcua in next release as it contains fixes.

TheeeEnding, 2024-04-12:
Thanks a lot @mikakaraila !
Just fyi, I've done a few tests in the meantime and was able to nail down the problem to the code line that I mentioned above.
With the current version a short downtime of the server is not a problem but when there is a longer downtime (e.g. 4h) the client cannot reconnect successfully. I did these tests also with a locally forked version where I commented the keepAlive line out and the reconnect was working even when there was a longer downtime. 
I hope the update will fix this issue! thank you!

mikakaraila, 2024-04-12:
I just made & published new version. Not fully tested, so I will fix if there is something...

TheeeEnding, 2024-04-16:
I tested the new version overnight in order to see the reconnect/keepAlive after a longer downtime and unfortunately the problem is not fixed!
The client did not reconnect successfully after the opcua server was down for a few hours

mikakaraila, 2024-04-16:
OK, what is server that client tries to reconnect? 
Can you see what happens when server comes back online?

I have to check node-opcua release notes and perhaps ask if Etienne has some experience...

mikakaraila, 2024-04-16:
I will take a look that keepAlive can be disabled...

TheeeEnding, 2024-04-16:
Thanks a lot!!

integratec, 2024-05-09:
I'm experiencing the same issue.
When the server goes down for long time the node continue to keepalive even if the server is unreachable and will never reconnect. The only workaround i found is by restarting the flow.
[image]

regrog, 2024-05-14:
> I'm experiencing the same issue. When the server goes down for long time the node continue to keepalive even if the server is unreachable and will never reconnect. The only workaround i found is by restarting the flow. ![immagine](https://private-user-images.githubusercontent.com/5618006/329344822-540e4350-4910-462e-a0fa-17d9dd3a5754.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU2NzQzNDcsIm5iZiI6MTcxNTY3NDA0NywicGF0aCI6Ii81NjE4MDA2LzMyOTM0NDgyMi01NDBlNDM1MC00OTEwLTQ2MmUtYTBmYS0xN2Q5ZGQzYTU3NTQucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDUxNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA1MTRUMDgwNzI3WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YTZlMWVkNzc3MzlkZDBhYWEwOTE2YzI1Njg1ODBiNGNmZWY0NDRhYzlmOWEyNzE1YTk5YTdiOGI2MDZlY2Y4MCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.kc0d3bL_mgUBCIqtGbebiECczSc1yBTSwg7lU6jKPH8)

Hi, take a look at this solution https://github.com/mikakaraila/node-red-contrib-opcua/issues/638

TheeeEnding, 2024-05-14:
@regrog Hi, how would that help? It seems that the solution from https://github.com/mikakaraila/node-red-contrib-opcua/issues/638 only works if the node throws an BadConnection Error. In this example the node does not show any errors but stays in the "keepalive" status

regrog, 2024-05-14:
I'm currently on version 0.2.328
I'm having long downtime during the night but yes my node throws BadConnection Error.
Did you try to upgrade to the last version?

TheeeEnding, 2024-05-14:
I've tested with the 0.2.327 and it still doesnt throw an error. The change from v0.2.327 to 0.2.328 only affects the opcua server node - so I didnt test it

integratec, 2024-05-14:
I'm on version 0.2.328, unfortunately no BadConnection errors raised. I already tried the solution at #638 but no luck, may be because the node still stuck in keepalive status like it's not aware the server isn't online anymore.

I think it should be considered normal/expected behaviour that a machine will no longer be accessible... this happens at least every day when a department ends its work.
I whould say thanks to @mikakaraila for it'work but sorry... I believe this issue should be labeled as a bug instead of enhancement ;-).

TheeeEnding, 2024-07-29:
Hey, can we expect any update on this? It is still an ongoing problem....
Thank you!

mikakaraila, 2024-07-29:
Sorry I am on vacation back to business after two weeks

MountainKing91, 2024-11-20:
Hi,
with version 0.2.337 I am experiencing the same issue; if the server/machine stays offline for a while (I'd say a few hours), the client will remain in "keepalive" state and will not receive any more data when the server actually comes back online. Currently the only solution seems to restart flows.

Have you guys found any other solution or workaround in the meanwhile?

TY