mikakaraila/node-red-contrib-opcua #678: get historical data when query given from any client to opcua server with start time and stop time

vnvasireddy, 2024-04-12:
hi,

i have used installHistorian for specific nodeid to save historical data, but if in any case server restarts the data saved is completely wiped off starts saving data from that point of time of restart.

Is there any solution that if any client requests for data with start time and stop time, opcua server will fetch data from database and publish it. 

Thanks in advance
vani

mikakaraila, 2024-04-12:
Sorry the current implementation is memory based ring buffer. I just have not had time to add interface for the database.

vnvasireddy, 2024-04-19:
ok..i will try from my side if you can help me in catching any **notification or event**  in server 
when requested from UAexpert tool or any other client.

Thanks in advance,
vani

mikakaraila, 2024-04-26:
Actually this should be implemented another way. Extending existing variable that will connect to database.
I have made implementation, but it is "legacy". Needs a bit more time to make it for general purpose.