mikakaraila/node-red-contrib-opcua #679: Using historian, read or read-history, adds an extra record to the history

peza, 2024-04-17:
Using historian, read or read-history, adds an extra record to the history.
I've been reading the server also with UAAutomation, where no extra records are added. Only in node-red via READ or HISTORY.
Attached FLOW for reproducing and a screeenshot for additional explanation.
[image]

[
    {
        "id": "fdb793e979dca26e",
        "type": "tab",
        "label": "BUG",
        "disabled": false,
        "info": "This flow shows the BUG, that adds an extra\r\nrecord to the historizing variable on read",
        "env": []
    },
    {
        "id": "8554c947a60928b6",
        "type": "OpcUa-Server",
        "z": "fdb793e979dca26e",
        "port": "4841",
        "name": "Test Server 4841",
        "endpoint": "",
        "users": "",
        "nodesetDir": "",
        "autoAcceptUnknownCertificate": false,
        "registerToDiscovery": false,
        "constructDefaultAddressSpace": false,
        "allowAnonymous": true,
        "endpointNone": true,
        "endpointSign": false,
        "endpointSignEncrypt": false,
        "endpointBasic128Rsa15": false,
        "endpointBasic256": false,
        "endpointBasic256Sha256": false,
        "maxNodesPerBrowse": 0,
        "maxNodesPerHistoryReadData": "0",
        "maxNodesPerHistoryReadEvents": "0",
        "maxNodesPerHistoryUpdateData": "0",
        "maxNodesPerRead": 0,
        "maxNodesPerWrite": 0,
        "maxNodesPerMethodCall": 0,
        "maxNodesPerRegisterNodes": 0,
        "maxNodesPerNodeManagement": 0,
        "maxMonitoredItemsPerCall": 0,
        "maxNodesPerHistoryUpdateEvents": "0",
        "maxNodesPerTranslateBrowsePathsToNodeIds": 0,
        "maxConnectionsPerEndpoint": "0",
        "maxMessageSize": "0",
        "maxBufferSize": "0",
        "maxSessions": "",
        "x": 1170,
        "y": 120,
        "wires": [
            []
        ]
    },
    {
        "id": "d18ee86b0f24dbda",
        "type": "inject",
        "z": "fdb793e979dca26e",
        "name": "opcuaCommand = installHistorian",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
[... 357 lines cut ...]
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 770,
        "y": 460,
        "wires": []
    },
    {
        "id": "71cb8fc28f03fd3a",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://localhost:4841",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]

mikakaraila, 2024-04-17:
If I remember correctly it return by default bounds... look specification. This might need some extra options...

PeterGostol, 2024-04-18:
Hi,
@mikakaraila why is this labeled enhancement? This is a bug. The extra record shouldt'n have been added after reading historian in node-red.

Let me explain again; When reading the variable via node-red, an extra record is added, which shouldn't been there. If i'd be using UAExpert, correctly, no additional record is added. And also, the extra record, added by node-red, is seen on the graph inside UAEXpert.

The extra record gets the value of the last record read and new timestamps. So, we get two records with same value, which is wrong!

mikakaraila, 2024-04-18:
I still think it is related to this:
https://reference.opcfoundation.org/Core/Part11/v104/docs/4.4

As node-opcua test coverage is very high I expect this is tested. I am using existing functionality.
So if you really think this is a bug then open issue in node-opcua.